claudegel/sinope-zha #9: Zigbee2qmtt support for Sinope.

chrisbklein, 2023-03-05:
Hello Claude.

Will there be ever an integration from you to support zigbee2mqtt for Sinope devices.
For all those people (like me) that do not like the idea of keeping all your information private instead of using a cloud app?

I do have several Sinope devices and use them without the Neviweb cloud based GT130

claudegel, 2023-03-05:
To use zigbee Sinope devices without the GT130 there are two possibility. Both require the use of a zigbee gateway like the conbee II
- ZHA, which is the one I'm doing some development. sinope-zha is the development part.
- Zigbee2mqtt which already support most of the Sinopé zigbee devices.
I'm doing development only for the first one, ZHA

claudegel, 2025-05-22:
To have the double _up, double _down etc work you need first to enable reporting on cluster 0xff01, attr 0x0054. Once done the device will report values depending on tap done on the up or down button. So you need to catch value from attr 0x0054. They are described on my readme.md section Light switch and dimmer double tap, long press reporting. At first try it was done via automation that catch the values from attr 0x0054 but now it is done directly in ZHA quirks by catching reporting events from attr 0x0054.
There should be a way to set reporting for attr 0x0054 in Z2M and it should be possible to catch reporting event. Then it should be easy to map the value reported from 0x0054 to each single, double or long press up or down.
I'm not a specialist of Z2M but I'll see if I can understand better Z2M.

sebjacob, 2025-05-25:
thank you for the insight @claudegel 

I've been trying to make it work in zigbee2mqtt myself, (I'd like to move all my dimmers to home assistant, leaving the thermostats on the gt130)

been using ai to do the research and test things, interesting stuff, if you're interested here's a "deep thought research" about porting the "quirks" : https://chatgpt.com/s/dr_6833a4c47bc881918852840fe35961d7

sebjacob, 2025-06-24:
# Real-Time Button Reporting for Sinope DM2550ZB in Zigbee2MQTT — Working Extension and Config

Hi all,

I wanted to share a **working and clean solution** for enabling **real-time button press reporting** from the Sinope DM2550ZB Zigbee dimmer in Zigbee2MQTT. This is built on the invaluable groundwork and insights by **Claude**, who was the true catalyst for understanding the manufacturer-specific cluster and attribute behavior of this device.

---

## Background / Problem

- The device sends button press events via the manufacturer-specific cluster `manuSpecificSinope` (0xFF01).
- The specific attribute that signals button presses is `0x0054` (decimal 84), called `actionReport`.
- While Claude's ZHA quirks demonstrated how to decode these actions rapidly, the equivalent support in Zigbee2MQTT was missing or incomplete.
- External converters encountered import and runtime issues. It broke the native support.
- Newest Zigbee2MQTT has a extension mechanism that, after a lot of tinkering works.
- Reporting settings were often set with too high minimum intervals, causing lag in event reporting.

---

## Key Contribution: Manual Reporting Configuration + Extension

After experimenting, I found that **manually enabling attribute reporting in Zigbee2MQTT** with a **minimum reporting interval of 1 second** for attribute `0x0054` is critical to get near-instantaneous event delivery.

This is done in the Zigbee2MQTT device configuration, for example:

```yaml
endpoint: 1
cluster: manuSpecificSinope
attribute: 0x0054
minimumReportInterval: 1
maximumReportInterval: 3600
reportableChange: 1
manufacturerCode: 0x119C
```

Either paste this in the extension section as a new file in the Zigbee2MQTT http page, or in `zigbee2mqtt/data/external_extensions/sinope_buttons.js`

```java
class SinopeDM2550Extension {
  constructor(
    zigbee,
    mqtt,
    state,
    publishEntityState,
    eventBus,
    enableDisableExtension,
    restartCallback,
    addExtension,
    settings,
    logger,
  ) {
    this.mqtt = mqtt;
    this.eventBus = eventBus;
    this.settings = settings;
    this.logger = logger;
    this.lastActionMap = {};
    this.mqttBaseTopic = this.settings.get().mqtt.base_topic;
    this.supportedModel = 'DM2550ZB';
    this.handle = this.handleDeviceMessage.bind(this);
    this.logger.info('Loaded SinopeDM2550Extension');
  }

  start() {
    this.logger.info('SinopeDM2550Extension started');
    this.eventBus.on('deviceMessage', this.handle, this.constructor.name);
  }

  stop() {
    this.eventBus.removeListeners(this.constructor.name);
    this.logger.info('SinopeDM2550Extension stopped');
  }

  handleDeviceMessage(data) {
    if (
      data.type !== 'attributeReport' ||
      data.cluster !== 'manuSpecificSinopeTechnologies' ||
      !data.data ||
      data.data[84] === undefined
    ) return;

    const attr = data.data[84];
    const dev = data.device;
    if (!dev.modelID?.includes(this.supportedModel)) return;

    const actionMap = {
      2: 'up_single',
      4: 'up_double',
      3: 'up_long',
      18: 'down_single',
      20: 'down_double',
      19: 'down_long',
    };

    const action = actionMap[attr];
    if (!action) return;

    const topic = `${this.mqttBaseTopic}/${dev.friendlyName}/action`;

    // Debounce duplicate actions within 300ms
    const now = Date.now();
    const key = `${dev.friendlyName}_${action}`;
    if (this.lastActionMap[key] && now - this.lastActionMap[key] < 300) return;
    this.lastActionMap[key] = now;

    this.logger.info(`📣 Sinope action: ${action} (0x0054=${attr}) from ${dev.friendlyName}`);
    this.mqtt.publish(topic, action);
  }
}

module.exports = SinopeDM2550Extension;

sebjacob, 2025-06-24:
@claudegel thanks for this ! it seems to work perfectly.  (hope this doesn't saturate the network, will monitor it)

all that remain would be the reporting of power and adaptive phase settings.

If by any chance you are aware if this is possible that would be great !

I think I can set that physically on the dimmer (I need trailing edge in multiple locations)