claudegel/sinope-zha #11: SW2500ZB & DM2500ZB

frangigu, 2023-06-17:
Hi Claude,

Not sure if is an issue and if it can be fixed, but I do have many Sinope Switches and Dimmers, reported device temperature is wrong.

All devices are showing 0.2C or 0.3C. It looks like the device_temperature value is divided by 10 while it should not be divided.
[image]

Not sure if this is Sinope that is not following the standard. Is this something that could be overwritten or changed in the switch.py ?

I am Java programmer but I know nothing about Python, not sure where to start and help ;( .

claudegel, 2023-06-17:
Should be easy to fix. I did the same type of correction for wall oulet that was reporting bad load value.
Let me check on mine. For light and dimmer the correction will be in light.py not switch.py

claudegel, 2023-06-17:
On my side device  temperature is 0.3 and in cluster 0x0002, attribute 0x0000 temperature = 31
So it should be multiplied by 100.
I'll try something

claudegel, 2023-06-17:
OK my dimmer and light are reporting correct temperature in the cluster 0x0002, attribute 0x0000
But ZHA is doing a division by 100. So the bug is in the ZHA code not in the sinope quirck. 
In the device temperature cluster 0x0002 there is no divisor or multiplier attribute so we cannot change there value to get the wright number for temperature.

I'll check in the ZHA code to find out where they get pr change the value for device temperature.

frangigu, 2023-06-17:
Thanks for the quick response and updates

frangigu, 2023-11-06:
I just noticed that numbers are now showing properly for my light switches. Not sure if it is after updating from your changes or a HA change.

claudegel, 2023-11-07:
I've fix it by adding *100 to the attribute value. All temperature are set as celcius*100 except for that attribute so I've made a little catch to transform the value as celcius*100 before it is used by ZHA

frangigu, 2023-11-07:
Thanks for the change

closed 2023-11-07 by frangigu