claudegel/sinope-zha #12: Question about supporting lm4110-zb

vladkvit, 2023-06-30:
Hi! I'm thinking of buying the lm4110-zb - the sinope Zigbee propane tank level monitor. 

I'm curious if you think it would be straightforward to add support for this device.

I'm a bit new to Zigbee with Home Assistant - do you know what the range is like? 

If I have a Home Assistant Yellow board (with onboard Zigbee 3.0), would that work (without using a gateway)?

claudegel, 2023-06-30:
Hi, I just received mine and already added support in my sinope-zha. The device will work with HA yellow as it have an integrated zigbee gateway. So I don't think that you will need to by another gateway.
As for the range mine is outside at about 20 feet of my gateway and it connect instantly with my gateway. because the zigbee devices work in a mesh network if the LM4110-ZB is far from the gateway it will automatically connect to another zigbee device that will serve as a relay. So it should not be a problem for the range.
I already have the outside temperature and battery voltage and percent remaining. I'm still working to get the tank level properly. I need to find the cluster that I need to poll to get the data. 
As I just received mine yesterday it should work soon in sinope-zha and also in my other component neviweb130 for thos who prefert to go via there GT130 and Neviweb.
Once it work perfectly I'll send a PR to have the quirk added into ZHA directly and I'll add more documentation in my readme.md
I hope it is ok for you.

vladkvit, 2023-07-11:
I bought one and successfully integrated it into HA using this repo.

For posterity, here's my automation code:

alias: Check propane level
description: ""
trigger:
  - platform: time_pattern
    minutes: /10
condition: []
action:
  - service: zha_toolkit.attr_read
    data:
      ieee: button.sinope_technologies_lm4110_zb_identify
      cluster: 12
      attribute: 85
      state_id: sensor.propane_tank_level
      allow_create: true
mode: single

claudegel, 2023-07-12:
I just sent a PR for the LM4110-ZB support to ZHA-device-handler.
For your automation, checking level every 10 minutes isn't a frequency too high. Once per hour should be enough ? Propane level do not change that fast.

vladkvit, 2023-07-12:
Thank you for sending the PR. And yep, 10 minutes was more for debugging purposes , just changed my script to hourly.

claudegel, 2023-07-12:
Thank you, I'm still working on a way to convert degree to % maybe with a template sensor of if I can, directly in ZHA.
I'll let you know once I have something working. I already have it working for my other custom-compoment neviweb130.
Problem is that there are two different gauge supported.

claudegel, 2023-08-27:
Hi, all the info about adding LM4110-ZB to zha, getting the gauge angle et converting it to % level is added in readme.md file.
I've added calculation for gauge 10-80 and gauge 5-95. Pick the one according to your tank gauge.

closed 2023-08-28 by claudegel

vladkvit, 2023-09-02:
Thank you. Just integrated the snippets from the readme. Some comments:
* I'm new to Home Assistant, so it took me a bit of time to figure out where the snippets should go (configuration.yaml; helpers), might make sense to add that.
* For the config, I adjusted it to provide default values as HA was complaining:
        state: >
          {% set gauge = states('input_text.propane_gauge') %}
          {% set angle = states('sensor.propane_tank_level') | float(0) %}
          {% set offset = states('input_number.propane_gauge_offset') | float(0) %}
          {% set x_min = 110 | float(0) %}
          {% if (gauge == "10-80") %}
            {% set x_max = 406 | float(0) %}
            {% set delta = 46 | float(0) %}
            {% set low = 10 | float(0) %}
            {% set high = 80 | float(0) %}
          {% else %}
            {% set x_max = 415 | float(0) %}
            {% set delta = 55 | float(0) %}
            {% set low = 5 | float(0) %}
            {% set high = 95 | float(0) %}
          {% endif %}

          {% if ((delta <= angle) and (70 >= angle)) %}
            {{ (((delta-x_min)/(x_max-x_min)*(high-low))+low+offset) | round(0) }}
          {% elif ((0 <= angle) and (delta > angle)) %}
            {{ (((angle)+360-x_min)/(x_max-x_min)*(high-low)+low+offset) | round(0) }}
          {% else %}
            {{ (((angle)-x_min)/(x_max-x_min)*(high-low)+low+offset) | round(0) }}
          {% endif %}

claudegel, 2023-09-02:
Yes the twp input_number and input_text goes in configuration.yaml
The name of those input must match in the template