claudegel/sinope-zha #13: Question inclusion ZHA DM2500ZB

karco, 2023-08-26:
Bonjour Claude, premièrement, merci pour tes efforts d'intégration Sinope.

J'ai un soucis avec l'ajout du DM2500ZB qui s'affiche dans HA mais qui ne peut être controlé.  Initialement, j'ai ajouté le DM2500ZB via ZHA sans problème.  Par la suite, j'ai voulu le tester via Neviweb et lorsque je suis revenu en arrière avec ZHA, HA voit le device mais ne peut communiquer les commandes.

Tu as déjà vu ce phénomène ?  Il y a une solution ?

Merci

claudegel, 2023-08-26:
Si neviweb a fait un update du firmware il est possible que l'appareil ne soit plus reconnu dans zha.
Peux-tu m'envoyer la signature de l'appareil dans zha ?
Il devrait être reconnu comme Quirk: light.SinopeDM2500ZB

karco, 2023-08-26:
Voici la signature du DM2500ZB

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4508, maximum_buffer_size=71, maximum_incoming_transfer_size=43, server_mask=10752, maximum_outgoing_transfer_size=43, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0702",
        "0x0b05",
        "0xff01"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Sinope Technologies",
  "model": "DM2500ZB",
  "class": "zhaquirks.sinope.light.SinopeDM2500ZB"

karco, 2023-08-26:
Je trouve ce message dans HA lorsque je tente d'intéragir avec la lumière

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1974, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2011, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 580, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 360, in async_turn_on
    result = await self._on_off_cluster_handler.on()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 75, in wrapper
    raise HomeAssistantError(message) from exc
homeassistant.exceptions.HomeAssistantError: Failed to send request: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>

karco, 2023-08-26:
Jusqu'à présent, j'ai fait plusieurs "reset", une changement de canal zigbee (vu que j'avais le message WARNING (MainThread) [zigpy.application] Zigbee channel 15 utilization is 99.91%!).

Si je remets le DM2500ZB dans Neviweb, tout fonctionne.

claudegel, 2023-08-26:
Je pense que ton problème vient d'interférence entre le zigbee et le usb3 ou le wifi. Pour corriger il faut prendre une rallonge de cable usb pour éloigner ton antenne zigbee du raspberrypi d'au moins 1 mètre. Ca va reduire les erreurs de communication zigbee et reduire ton utilisation du canal15. Tu dois avoir plein de drop de packets.

karco, 2023-08-26:
Mon installation a déjà une rallonge pour éloigner la communication.  Aussi, j'ai fermé les autres émetteurs d'ondes (zwave, wifi, bluetooth et les autres hub zigbee) et je n'ai pas plus de chance.

Je viens de tenter d'inclure deux autres appareil Sonoff zigbee (Température/Humidité) et tout fonctionne correctement.

karco, 2023-08-26:
Je viens de faire la même manipulation avec un SW2500ZB et c'est le même symptome.

- Ajout/suppression dans neviweb
- Ajout dans HA sans succès.  ZHA trouve le device et lors de la configuration, il perd le lien avec le device.

Je ne sais plus où regarder.

karco, 2023-08-26:
Nouveau test avec une autre DM2500ZB sans faire l'étape neviweb.  J'ai pu ajouter de nouveau dans HA sans problème.

karco, 2023-08-26:
Nouveau test avec une autre DM2500ZB sans faire l'étape neviweb. J'ai le même problème dans HA lors de la réinclusion.

claudegel, 2023-08-27:
Combien tu as de device connecter à ton gateway zigbee et quel est ce gateway ?

karco, 2023-08-27:
J'ai 37 devices, incluant les trois Sinope non fonctionnels, sur un SONOFF ZigBee 3.0 (EFR32MG21)

claudegel, 2023-08-27:
Ca roule basé sur Silicon Labs EmberZNet et il y a pas mal de bug si tu surcharges le réseau. Tu as déjà un message d'erreur qui dit Zigbee channel 15 utilization is 99.91%. Je pense que le bug viens de ton Sonoff.
De mon coté j'ai un conbee II avec 30 devices qui fonctionnent. J'ai eu des problèmes avec des xiaomi.aqara.motion_aq2 qui sont réputé pour perdre le lien avec le gateway. mais tous mes Sinopé fonctionnent très bien. Un certain temps j'en avais plus mais j'en ai retourné sur Neviweb pour terminer mon développement.

karco, 2023-08-27:
Je vais tester en me branchant un deuxième EFR32MG21.  Je te reviens avec le résultat bientôt.

Merci pour ton aide

karco, 2023-08-27:
J'ai complété l'intégration du deuxième EFR32MG21 via Z2M et je réussis à ajouter l'intérupteur DM2500ZB sans problème.

claudegel, 2023-08-27:
Es-tu sur le même canal que ton autre  EFR32MG21. De toute façon si les dimmer marchent dans Neviweb et dans z2m alors ce ne sont pas les dimmer qui sont en problème. Puisque mes dimmers marchent sans probleme sur zha avec mon conbee II ça doit être ton premier EFR32MG21 qu est défectueux ou qui perd le contrôle quand il y a plein de trafique sur le canal 15.
Faudrait que tu install ton deuxième EFR32MG21 sur zha en y ajoutant tes appareils un par un en commençant par les dimmer pour valider que c'était bien ton EFR32MG21 qui en perd des bout.

J'ai eu un cas ou des thermostats wifi se débranchaient constamment de neviweb et le problème venait du router qui ne peu supporter plus de 15 appareils wifi connectés en même temps. Ajouter une passerelle wifi a réglé le problème.