claudegel/sinope-zha #17: TH1123ZB config_2nd_display enum

arno1801, 2023-10-29:
I noticed that the enumerations 0x01 & 0x02 for 2nd display with TH1123ZB & TH1123ZB-G2 are swapped. The good order is
1 = outside
2 = set point.

claudegel, 2023-10-29:
in thermostat.py I have 
```
    class Display(t.enum8):
        """config_2nd_display values."""

        Auto = 0x00
        Outside_temperature = 0x01
        Setpoint = 0x02
```
Where did you see the error ?

claudegel, 2023-10-29:
I think it should be 
```
    class Display(t.enum8):
        """config_2nd_display values."""

        Auto = 0x00
        Setpoint = 0x01
        Outside_temperature = 0x02
```

arno1801, 2023-10-31:
I'm sorry, it's should be like you said 

> I think it should be
> 
> ```
>     class Display(t.enum8):
>         """config_2nd_display values."""
> 
>         Auto = 0x00
>         Setpoint = 0x01
>         Outside_temperature = 0x02
> ```

Sorry, I mixed the values in my first post. When I send 1, the thermostat display the actual set point and when I send 2, it display outside temperature. So it should be like you said in your second post.

claudegel, 2023-10-31:
Ok thank you for reporting I've fix it in the thermostat.py file
Auto = 0x00
Setpoint = 0x01
Outside_temperature = 0x02

closed 2023-10-31 by claudegel