claudegel/sinope-zha #18: TH1300ZB current_load cluster: 0xff01 - 0x0070

frangigu, 2023-11-06:
Hi Claude,

I see an issue with my floor Thermostat TH1300ZB. The current load showing in device attribute is always set to the "connected_load" value no matter what level it is heating.

If I look in HA I always see 528W which is my max even if the Thermostat has 1, 2, 3, 4 or 5 lines. When it turns off after hitting setpoint it goes back to 0.

I have been using your quirks since end of 2022, last winter everything was fine and I could see the proper level showing. I haven't put it back in the Sinope Neviweb environment since then, so my firmware has not been updated since I moved to HA.

Could it be one of the multiple changes you have made in thermostat.py since last winter ?

claudegel, 2023-11-07:
Hi on the thermostat the 5 horizontal bar are related to pi_heating_demand which is expressed in % .
Not sure to understand does your thermostat is in zha-sinopé or neviweb130. Where do you see current load in device attribute ?

claudegel, 2023-11-07:
In ZHA energy is expressed as «va» witch is equivalent to «w» or KVA = KW but I don't see where it is directly expressed as 528W
It is the Electricalmeasurementapparentpower and that value change as the heating demand increase and it is expressed as for example 312 VA which is equivalent to 312 W
Can you show me screen capture
[image]

frangigu, 2023-11-07:
As you can see, the power which should variate is alway at max 528w while
my thermostat shows 2 horizontal bar... Should be abot 40% of 528

claudegel, 2023-11-07:
Sorry there is no picture

claudegel, 2023-11-07:
As you can see on the graph the orange color under the curve is the heating . when the setpoint is raised, the orange fill all the space under the line which is when you have 5 bars on the thermostat. then as temperature increase and is close to the desired temperature, the thermostats will start to pulse the heating from 0 to 100%. this is the vertical orange line on the graph. 
In the first part, the VA measured if it last for an hour is equal to the connected load in kwh. but as the thermostat start to pulse then the energy is lest then the connected load. but to see the level you need to multiply the connected load by the pi heating demand in %

[image]

frangigu, 2023-11-07:
Heres is an exemple for a TH1124ZB : 
[image]
[image]
*See "Power" sensor, it gives real time consumption in watts. This thermostat has a load of 3900w as per Thermostat reading even in real life it is about 3750w

This is what happens on my TH1300ZB Floor :
[image]
[image]
 38% of 504 does not equals 504
Sensor Power always reports 100% consumption, but it is not true

frangigu, 2023-11-07:
Only floor thermostat seems to be affected

claudegel, 2023-11-07:
I see. I've check on TH1123ZB and it is reporting correctly. My TH1300zb show the same as you but I have only the apparent power  the other power is not there. What is the sensor name when you clic on power.
[image]

frangigu, 2023-11-07:
It is sensor.thermostat_plancher_salle_de_bain_active_power

This is strange because last year it worked perfectly as I put it into many templates to monitor real time household usage

claudegel, 2023-11-07:
The sensor are created by ZHA so maybe it is in ZHA that they pick a different cluster/attribute.
to get the value you need, you can take 
(sensor.thermostat_plancher_salle_de_bain_active_power / 100) * pi heating demand.
Do you have zha toolkit installed ? with that it is possible to view all cluster attributes of your thermostats with a single service
Normally ZHA is creating sensor from cluster/attribute that are doing reporting. by comparing both thermostat cluster it will be possible to find out what is different and how to change it. Like what I did for the light temperature that was reported as 0.3 instead of 30 oC

claudegel, 2023-11-07:
On my TH1123ZB-G2 it work perfectly. the power increase as the pi heating demand increase

claudegel, 2023-11-07:
Install zha toolkit via HACS and run that service 
```
service: zha_toolkit.scan_device
data:
  ieee: 50:0b:91:40:00:03:ed:b0 <-- ieee of your thermostat
  endpoint: 1
  event_success: my_read_success_trigger_event
  event_fail: my_read_fail_trigger_event
  event_done: my_read_done_trigger_event

```
This will create a file in config/scans. scan your TH1124ZB and your TH1300ZB so I'll be able to compare cluster and find where ZHA take the value for power  and apparent power

frangigu, 2023-11-08:
Here are the scans
[TH1124ZB_Sinope-Technologies_4000025d0a_01_scan_results.txt](https://github.com/claudegel/sinope-zha/files/13290460/TH1124ZB_Sinope-Technologies_4000025d0a_01_scan_results.txt)
[TH1300ZB_Sinope-Technologies_4000021327_01_scan_results.txt](https://github.com/claudegel/sinope-zha/files/13290461/TH1300ZB_Sinope-Technologies_4000021327_01_scan_results.txt)

claudegel, 2023-11-08:
Thank you I've received your two files. Strangely TH1300ZB do not show the cluster attribute for 0xff01 which is the manufacturer cluster for the thermostat. TH1124ZB is ok and show the cluster. This is normally due to Sinopé that didn't activate teh view for that cluster and this don't mean that it is not there. I have many dimmer with the same problem.
I'll check how to read the attributes that we need and come back to you soon.

frangigu, 2023-11-08:
Ok, thanks but I am not sure I understand, it seems like the cluster is there and readable 
[image]

claudegel, 2023-11-08:
Yes but the command scan_device is not able to read it so we will revert to reading attribute one by one like this
```
service: zha_toolkit.attr_read
data:
  ieee: 50:0b:91:40:00:01:f1:e7   <---ieee of your TH1300ZB
  endpoint: 1
  cluster: 0xff01
  attribute: 0x0070
  event_success: my_read_success_trigger_event
  event_fail: my_read_fail_trigger_event
  event_done: my_read_done_trigger_event
```

try attribute 0x0070, 0x0119

But you can also use the method via zha as you did above

claudegel, 2023-11-08:
can you tell me what is teh value for cluster 0xff01 attribute 0x0000

frangigu, 2023-11-08:
zha_toolkit in txt files
[0xff01 0x0070.txt](https://github.com/claudegel/sinope-zha/files/13291252/0xff01.0x0070.txt)
[0xff01 0x0119.txt](https://github.com/claudegel/sinope-zha/files/13291253/0xff01.0x0119.txt)
[0xff01 0x0001.txt](https://github.com/claudegel/sinope-zha/files/13291254/0xff01.0x0001.txt)
[image]
[image]

I cannot see 0x0001 attribute in 0xff01 cluster from the UI/zha

claudegel, 2023-11-08:
Check this way
```
service: zha_toolkit.attr_read
data:
  ieee: 50:0b:91:40:00:02:13:27
  endpoint: 1
  cluster: 0xff01
  attribute: 0x0001
  event_success: my_read_success_trigger_event
  event_fail: my_read_fail_trigger_event
  event_done: my_read_done_trigger_event
```
On the same page you will see many lines. Check the last three:
    Result: {257: <Status.UNSUPPORTED_ATTRIBUTE: 134>} - Attribute 257 not in
    read ({}, {257: <Status.UNSUPPORTED_ATTRIBUTE: 134>})
success: false

The attribute is not there

- "1": 0
  - {}
success: true
The attribute is there with value 1

frangigu, 2023-11-08:
Yes, I saw it in zha toolkit attr read, but just couldn't find from UI

claudegel, 2023-11-08:
Ok got them. For attribute 0x0070 I think it is apparent power for your Th1300ZB.
In zha for this thermostat if you click  on the apparent power line you should see the values history. Is it only 0 or 504 or you have a variation ?

frangigu, 2023-11-08:
Apparent power is in cluster 0x0b04 and is always 504 event when off, it is the equivalent of connected load

As for current load which is actually 0x0070 it is 0 or 504 no variation even at minimal heat

claudegel, 2023-11-08:
Both attributes are the same. Sinopé use it's own and the thermostat write the value in 0x0070 and in cluster 0x0b04
We need to find out why your thermostat report only 0 or 504. maybe a defective thermostat. How old is it ?

frangigu, 2023-11-08:
About two years I think

frangigu, 2023-11-08:
How is your reporting? Do you see variation or not?

claudegel, 2023-11-08:
In the scan of your TH1300 I see cluster 0x0b04 attr 0x050f apparent power which is at 504
But there is also the attribute 0x050b active power which is at  0
On my TH1300 0x050b is not there and my reporting is ok. It's on 0x050f
I'll check what is the reporting frequency. maybe we can increase it soit report as soon as there is a change on the value.

frangigu, 2023-11-08:
I am not sure I understand what you just said, I was wondering if your "active_power" for your floor thermostat was reporting the maximum value or 0 no matter what the level is...

Anyway, since pi_heating_demand seems to be working properly and is in the attributes of thermostat from zha I will change my templates so it used to calculate current load and disable sensors that are not reporting correct values.

I was previously calculating the percentage using active_power/active_power_max*100, really seems like a breaking change in ZHA

I also don't think my device is defective, it seems to work properly on every aspect except the current_load/active_power that no longer report the proper value

claudegel, 2023-11-25:
I've talked to Sinopé and they told me that floor thermostats have a relay that can be all on or off so there is no intermediate value for active power. the pi heating demand is the frequency that this relay is turned on and off. So to get the active_power we need to multiply active_power_max  * pi_heating _demend %

frangigu, 2023-11-26:
Thanks, did it for my floor thermostat... just so you know, pi_heating_demand is updated immediately when it starts and then updates every 15 seconds from my observation. So I use this only for floor thermostats

closed 2023-11-26 by frangigu