claudegel/sinope-zha #20: Issue with missing attributes cluster 0xff01

frangigu, 2023-11-25:
Hi Claude, following my other issue, it seems like there is a bigger problem with thermostat.py

I am missing many attributes listed into your README.md, none of the ecomode (1/2/3) are available for me.

I have automation using attribute ecoMode1 (0x0072) 114, they no longer work.
[image]

Thats all I have in 0xff01 for every single thermostat I have (10 in total, 2x TH1124ZB 7x TH1123ZB and 1xTH1300ZB)
[image]
[image]

frangigu, 2023-11-25:
Forget this for now, I was on previous commit, updating to your last week commit seems to have solved the issue, attributes are back

closed 2023-11-25 by frangigu

claudegel, 2023-11-25:
I will have a bigger commit soon. In fact I've added 4 PR to ZHA which have been merged in dev branch and I'm waiting for the next ZHA release to activate them in sinope-zha.

frangigu, 2023-11-25:
Should I update firmwares of my different devices? I have over 30 sinope product all on zha for about 12 to 13 month?

I would like to avoid having to switch them all on Neviweb for a few days to get them updated... Do you keep any trace of firmware version I can compare to? Is it worth updating anyway ?

claudegel, 2023-11-25:
This is what I have for the device I have in Neviweb now:
TH1123ZB-G2: 0.1.1
TH1300ZB: 1.1.9
TH1124ZB-G2: 0.1.1
RM3250ZB: 1.3.3
RM3500ZB: 0.3.0
VA4200ZB: 1.3.2
VA4220ZB 2n gen with flow meter: 0.1.4
LM4110-ZB: 0.1.1
MC3100ZB: 0.1.1
DM2550ZB: 0.1.2
DM2500ZB: 1.3.2