claudegel/sinope-zha #21: Calypso - ON/OFF Toggle doesn't work after power outage

glavoie, 2023-11-28:
Hello,
     I just installed a Calypso yesterday and connected it directly to ZHA. Tested the power outage/power cycle scenario and the Calypso starts in the off state. Just after it gets power back, it does connect back through ZHA (Link) and reports its status, but I cannot switch its state through the HA ON/OFF toggle widget. I need to press on the physical ON/OFF button once to enable the Load, then in HA the ON/OFF widget will work to switch it back off, and back on again. 

Is there a way to make it enable the Load automatically and to make the ON/OFF toggle work in HA after the Calpyso gets powered again? 

Thanks!

claudegel, 2023-11-29:
Bonjour,

J'ai testé avec le miens et c'est un délais de 10-12 minutes puis il retourne à «on».
Il faut donc trouver un attribut qui a une valeur d'environ 720 soit 720 secondes avant de retourner à sa valeur d'avant la panne.

glavoie, 2023-11-29:
J'ai fait aussi quelques tests ce soir et en effet il semble se rallumer seul après une certaine période. Ça semble aléatoire par contre. J'ai eu entre 1 et 10-12 minutes. 

J'ai le même comportement avec Zigbee2MQTT. Soit c'est un glitch avec HomeAssistant, soit c'est un mécanisme pour ne pas surcharger le circuit avec des délai aléatoire de redémarrage sur leurs appareils. 

Tant que ça repart tout seul je suis content.

Merci des tests de ton côté.

claudegel, 2023-11-29:
J'ai testé plusieurs attributs qui sont encore non identifiés pour voir si ça aurait un impact sur le délais mais ça ne donne rien a date. J'ai demandé à Sinopé et on verra sa réponse demain. par contre je serai loin de mon ordi jusqu'à vendredi

glavoie, 2023-11-29:
Pas de trouble, je vais attendre que tu reçoives une réponse de leur part.

claudegel, 2023-12-08:
Finalement il y a même un message dans Neviweb lors d'un retour de panne. Neviweb applique un délais aléatoire pouvant aller jusqu'à dix minutes pour éviter que tout redémarre en même temps ce qui pourrait affecter les appareils. je vais tester avec d'autres type d'appareils pour voir ce que ça donne