claudegel/sinope-zha #26: Outside temperature

michaeldufour, 2024-01-17:
Bonjour Claude merci pour ton code ZHA ça fonctionne très bien.
J'ai créé une automation pour pousser la température extérieur sur mes thermostats j'ai pris un capteur de température intérieur que j'avais pour faire le test. De ton coté quel type de capteur extérieur que tu utilise qui est capable de lire des températures de jusqu'à -30 dans l'idéal en Zigbee?

Merci

claudegel, 2024-01-17:
Bonjour, j'utilises un MC3100ZB avec une sonde de température. Le MC3100 a une sortie pour une sonde extérieure et je pass le fil de la sonde par le tuyau d'entrée d'air de mon échangeur d'air. Donc pas besoin de percer pour sortir le fil.  je pense qu'elle va de -40 à +40

michaeldufour, 2024-01-17:
Merci de l'info Claude, je vais me procurer ce module ça sera parfait, petite dernière question est-ce possible de pousser la température sur les thermostat Sinope WIFI?

michaeldufour, 2024-01-18:
Bonjour merci Claude pour le retour, en fait j'ai commencé à installer du Zigbee et je pense peut-être à revendre les WIFI pour les remplacer :)

claudegel, 2024-01-19:
C'est de valeur qu'on n'est pas de controle local sur les wifi.  On l'a sur les miwi et les zigbee. je pense que je vais en trouver un pour pouvoir le tester en local en ecoutant le trafic wifi entre le thermostat et neviweb.