claudegel/sinope-zha #32: firmware version.md hex values

frangigu, 2024-01-28:
Do you have the equivalent value that is shown in HA in hex format or how to convert it in order to know if it is out of date prior to switch and wait a firmware update ?

Also, I remember my first devices could take a while before being updated and it was done at night time. If I switch a device for an update from HA to NeviWeb, do you have an idea of how long it could take to receive the firmware update ?

claudegel, 2024-01-28:
Normally the firmware update in neviweb is performed within 24 hours.
If you are in ZHA the firmware is written like this in parameters/devices and services / ZHA. Pick one devices and check in left column for this:
Micrologiciel : 0x01020902 = 1.2.9 (01 02 09)
I've put a file listing all device latest firmware in Neviweb, firmware version.md If you have other devices in neviweb that are not listed there please send me the firmware number so other can see and compare with what they have in HA.

frangigu, 2024-01-28:
While this methodology seems to work for some devices, it is not working for some other devices :

[image]

[image]

In this one it seems that 02 == HW Revision and 44 == 68 which can end up as 0.6.8 (I guess)

Your example seems to only work with light buttons...
[image]
[image]

I was asking for this because I have 36 devices in ZHA and I don't really feel like switching everything for 24 hours if not necessary because all my devices have automations and stats linked to them.

claudegel, 2024-01-28:
That's why I've added a firmware list so other can see where they are and if they need to switch back to Neviweb to have there firmware updated.
We need to find a better way to know the FW of our devices in ZHA.There is the cluster 0xff01, manufacturer cluster with attribute 0x0004 which is the firmware version but I don't know how to decode this as it is a number like 1269 and the FW is Micrologiciel : 0x01020001. It's a TH1300ZB and the FW is 1.2.0 in Neviweb. I don't know if there is a way to convert 1269 to 1.2.0

claudegel, 2024-01-28:
I'll check where ZHA take those number. Maybe there is a bug there to decode the firmware number.