claudegel/sinope-zha #33: unoccupied_heating_setpoint

michaeldufour, 2024-01-30:
Bonjour Claude

Je tente de pousser une température pour que lorsque que mes thermostats sont en mode "Away" il puisse être à 15 Degré Celsius

Est-ce qu'il y a quelque chose de pas correct dans mon script car impossible de changer la valeur actuel

service: zha.set_zigbee_cluster_attribute
data:
  ieee: bc:02:6e:ff:fe:ba:27:f7
  endpoint_id: 1
  cluster_id: 513
  cluster_type: in
  attribute: 20
  value: 15*100

claudegel, 2024-01-30:
La valeur a transmettre est celsius*100, donc 15oC c'est 1500 et non 15*100

claudegel, 2024-01-30:
Dans zigbee toutes les valeurs de température doivent être multipliées par 100 avant de les transmettre aux appareils pour utiliser des nombres entiers. Les valeurs recu des appareils sont automatiquement divisées par 100.

closed 2024-01-30 by claudegel