claudegel/sinope-zha #35: TH1123ZB-G2 False reporting Off

slooney38, 2024-03-13:
I am experiencing a problem where my TH1123ZB-G2s are false reporting their OFF states.  See attached photos indicating the the thermostat is off while further inspection under device info shows HVAC mode as heating (heat confirmed as on).
I have 1 TH1123ZB where this is not an issues, only effecting the G2 model.  Firmware version 0x00000900, HA 2024.3.0

Please let me know if you need any additional information.  Thank you!

[image]
[image]
[image]

slooney38, 2024-03-13:
I think I may be the problem.  I disconnected one of my G2s from the network, restarted HA and reconnected.  Looking under "manage zigbee device". the cluster attributes look like they have been properly updated to suite the G2.  I'll try to duplicate with my other G2s and close the issue is successful.  Thank you!

slooney38, 2024-03-15:
I've noticed that the error occurs when the heat is manually set from the unit with the G2s.  If I manually set the heat from my lone TH1123ZB is will correctly display as heating in home assistant.

claudegel, 2024-03-31:
Could you listen to the events or check in the journal of events for your G2 thermostat and switch the heat level manually on the device to see if you receive something like a report from the thermostat. Maybe is it just the reporting that is not set on your device.

slooney38, 2024-03-31:
I hope this is useful information as I am not familiar with using "listen to events"
1) Disabled temporary fix automation
2) Manually increased heat
[image]
 on G2 model
[image]
I can see the HVAC action change to heating the thermostat entity state remains "off"

Performing the same test with my non-G2 will toggle the thermostat entity state to "heating"