claudegel/sinope-zha #42: Ambient Light Sensor?

ELind77, 2024-05-20:
Hi Claude,

I'm looking for new thermostats for a large collection of baseboard heaters and I've been investigating the Sinopé Zigbee models as the best bet for local control with Home Assistant.  I was browsing the [marketing material for the TH1123ZB](https://www.sinopetech.com/en/products/thermostat/line-voltage/smart-thermostat-for-electric-heating-zigbee-th1123zb-th1124zb/) and I saw this:

> Adaptive backlit display harmonizes the screen brightness with the light intensity of each room

But I don't see an ambient light sensor in any of the sensor values in your excellent readme.  Do you think it actually has some kind of light sensor? Maybe one of the unknown values (if it were me, I'd use a `uint8`)?  Or maybe it's just marketing copy for the on-demand dimming? 

I also couldn't find a description for what the "Bedroom" dimming policy is but it is not clear to me if that is related. 

In any case, thank you for all of your hard work on these devices, your work might just make the decision of what thermostats I buy.

claudegel, 2024-05-21:
Hi, the bedroom dimming is an option of the backlight intensity. By choosing bedroom option the thermostats backlight will turn off when the light intensity of the room is low and turn on again during the day. It work perfectly.
just set the cluster 0x0201, attribute 0x0402,	BacklightAutoDimParam  to value 2. Possible value are:
OnDemand: 0, Always on: 1, bedroom: 2
You can do it via an automation via service: zha.set_zigbee_cluster_attribute. Let me know if you need help creating your automation.
You can also go to parameters, device and services, zha, select your thermostat and click on the three vertical dots (reconfigure), manage zigbee device. Then select cluster 0x0201, attribute 0x0402. Then put 2 as value and clicwrite attribute.

ELind77, 2024-05-21:
Thank you for explaining that, but how does it know when the light intensity of the room goes up or down? Does it have a light sensor or photo diode of some kind?

claudegel, 2024-05-21:
Yes there is a sensor in them. I've done a test on mine. Now the display is off but if I open the ligth the display turn on and if I turn off the light, the display turn back off

ELind77, 2024-05-21:
Do you know if we can read the value of that light sensor? That could be a potentially useful feature.

claudegel, 2024-05-21:
Upto now no, but you gave me an idea so I'll chek some unknown attributes to see if they can be related to light sensor. Could be light level or just an on/off attribute.
I'm gonna sniff the zigbee traffic to see if I can find an attribute value that change as the light intensity change. If I find something I'll add it to the cluster/attribute list and make it available in ZHA

ELind77, 2024-05-21:
That's great! Thank you. I think there may be opportunities for people to use a light sensor value for things like presence detection and other automations.