claudegel/sinope-zha #43: Intgration des HP600ZB

bsauve, 2024-05-23:
Bonjour Claude,
je vois dans neviweb130 que 2 controleurs HP6000ZB sont disponible.
Est-ce qu'ils le sont aussi dans ZHA ? 
Je ne les vois pas dans la liste de sinope-ZHA.

Sinopé m'ont dit que la version pour Thermopompe Midea sera disponible, à un moment donné.

Merci
Benot

bsauve, 2024-05-23:
Merci,
mais une autre question, quand on utilise les thermostats de pleinte ou de plancher Sinopé avec Zha et non avec neviweb130, est-ce que nous avons la même fonctionnalité de programmation par Home Assistant ?

Merci

claudegel, 2024-05-24:
Oui on a la même fonctionnalité mais en plus c'est plus rapide car on parle directement au thermostat et il envoi ses notifications directement à HA.
tu peux même combiner des appareils de zha avec des appareils de neviweb130 dans la même automation. Ex.

```
- service: climate.set_temperature
        entity_id:
          - climate.salle_de_bain_thermostat   <--- zha
          - climate.neviweb130_climate_chambre_1   <--- neviweb130
       ...
```

JeanOlivier, 2024-10-27:
Bonjour @claudegel,

J'ai reçu deux dongles HP600ZB et j'en ai connecté un sur mon réseau ZigBee sans problème (l'autre sera intégré plus tard). Je n'ai pas le gateway Sinope mais bien juste un adaptateur sur mon ordi avec HA.

Pour l'instant, j'ai des fonctionnalités très limitées pour la thermopompe, mais de ce que je comprend tes outils permettraient d'améliorer la situation? 

Merci de me diriger vers l'information pertinente, il ne semble pas y avoir des tonnes de gens avec des HP600ZB. Je suis aussi prêt à faire des tests pour aider à l'intégration.

Merci!

claudegel, 2024-10-27:
Bonjour @JeanOlivier as-tu la signature de ton HP6000ZB pour voir comment ZHA le détecte.
Vas dans parametres/Appareils et services/zha. Sélectionne ton appareil HP6000ZB. clic sur les trois petits points verticale a coté de reconfigurer et dans le menu clic sur Gérer l'appareil Zigbee.
Ensuite selectionne le troisième onglet, signature et publis ici ce qui s'y trouve
[image]

JeanOlivier, 2024-10-27:
Voilà:

``` yaml
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 140,
    "manufacturer_code": 4508,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0307",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0008",
        "0x0201",
        "0x0202",
        "0x0204",
        "0x0402",
        "0x0b05",
        "0xff01"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0307",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0008",
        "0x0201",
        "0x0202",
        "0x0204",
        "0x0402",
        "0x0b05",
        "0xff01"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Sinope Technologies",
  "model": "HP6000ZB-MA",
  "class": "zigpy.device.Device"
}
```

claudegel, 2024-10-27:
Parfait je vais faire le quirck pour le HP6000ZB-MA. As-tu installé mon module sinope-zha ? Ça permet de tester plus facilement. Je vais créer une branche pour que tu puisse tester avec ton HP6000ZB-MA.
Je t'avise quand cèst pret

JeanOlivier, 2024-10-27:
Non je n'ai pas sinope-zha. Je vais tenter de regarder ça quand j'aurai le temps. Si tu as un lien je suis preneur. 

Merci beaucoup!

claudegel, 2024-10-27:
https://github.com/claudegel/sinope-zha

claudegel, 2024-10-28:
J'ai créé le quirck et je test pour m'asurer qu'il load dans HA. Je vais créer une branche dans sinope-zha pour que tu puisses le tester.
L'étape suivante sera de trouver les cluster et attribut spécifiques à ton HP6000ZB-MA
Pour ça il faut installer le package zha-toolkit via HACS. Ce package permet de domper tous les cluster et attributs de ton thermostat ce qui va me permettre de paufiner le quirck.

claudegel, 2024-10-28:
Bon voilà. J'ai créé la branche HP6000ZB
https://github.com/claudegel/sinope-zha/tree/HP6000ZB
une fois que tu as installé sinopé-zha, remplace le fichier thermostat.py par celui de la branche HP6000ZB.
Tu peux ajouter dans configuration.yaml un peu de débug:
```
logger:
  default: warning
  logs:
    homeassistant.custom_components: debug
    custom_components.neviweb130: debug
    homeassistant.components.zha: debug
    zhaquirks: debug
```
Redémarres HA et ton HP6000ZB-MA devrait être reconnu dans zha comme 
"class": "thermostat.SinopeHPThermostats"

JeanOlivier, 2024-10-28:
Merci beaucoup Claude, je vais essayer de trouver le temps de faire ça dans les prochains jours.

claudegel, 2024-10-29:
Pas de problème. laisse moisavoir si tu as besoin d'aide. Sinope-zha ça remplace les quircks de zha avec les même fonctionalités. Quand ZHA démarre il détecte les modules de sinope-zha et remplace ses modules officiel par ceux de sinope-zha.
As-tu un GT130 ? Si oui j'aimerais savpoir quel est le numéro de model quand ton HP6000ZB-MA est connecté à Neviweb. Possiblement aussi quelques parametres pour raffiner le support du HP6000ZB-MA sur mon custom_component neviweb130 pour HA,

JeanOlivier, 2024-10-29:
Je n'ai pas le GT130, désolé.

claudegel, 2024-10-29:
Pas grave j'ai un bon bout de fait et je voulais juste valider le numéro de model qui Sinopé donne au HP6000ZB-MA

JeanOlivier, 2024-11-03:
Salut Claude, 

Je crois avoir tout installé comme il faut, la thermopompe est maintenant détectée comme: 

``` yaml
  "manufacturer": "Sinope Technologies",
  "model": "HP6000ZB-MA",
  "class": "thermostat.SinopeHPThermostats"
```
quelle est la suite?

claudegel, 2024-11-03:
C'est parfait, ton HP6000ZB-MA est correctement détecté. Il reste juste a scanner les cluster de ton appareil avec ZHA Toolkit. Commence par installer ZHA Toolkit dans HACS ensuite je vais t'envoyer une patch car il y a un bug qui enpèche de scanner les appareils. J'ai soumis la correction mais c'est pas encore corrigé dans ZHA Toolkit.

JeanOlivier, 2024-11-03:
C'est déjà installé, merci de forward le patch!

claudegel, 2024-11-03:
la commande est la suivante:
```
action: zha_toolkit.scan_device
data:
  ieee: 00:0d:6f:00:05:7d:2d:34  <--- le ieee de ton HP6000ZB
  endpoint: 1
  event_success: my_read_success_trigger_event
  event_fail: my_read_fail_trigger_event
  event_done: my_read_done_trigger_event
```
On fait ça dans outils de développement / actions. ça prends un certain temps puis affiche les résultats du scan dans la même fenêtre. Ca produit aussi un fichier dans config/scans avec le nom de ton appareil et tout le contenu des clusters, incluant celui de Sinopé 0xff01. Il va falloir répéter le scan pour le endpoint 2.
Je vais t'envoyer les modif a faire dans certains fichiers de config/custom_components/zha-toolkit

claudegel, 2024-11-04:
Vas sur https://github.com/mdeweerd/zha-toolkit/pull/263 il y a 4 liens pour modifier zha toolkit
@michalmo
https://github.com/mdeweerd/zha-toolkit/pull/263/commits/2cb2e02c6830e42109e1fae34a29f80bdbbcc53d
@michalmo
https://github.com/mdeweerd/zha-toolkit/pull/263/commits/f27fa190bae773b510f310c59367aedd42d58a34
@michalmo
https://github.com/mdeweerd/zha-toolkit/pull/263/commits/55f440af55f3b28cb37ed7acb396e14bfa5e72c0
@michalmo
https://github.com/mdeweerd/zha-toolkit/pull/263/commits/4f0b657138399e1a567f0dd8e9888d6f538fd15a

En gros il faut éditer les fichiers `__init__.py`, scan_device.py, utils.py, zcl_attr.py

claudegel, 2024-11-11:
J'ai poussé un PR dans zha device-handler pour le support de HP6000ZB-xx et j'ai ajouter la branche dans le master de sinopé-zha.
Si tu réussis à scanner les cluster ça permettrait de finaliser le support
Laisses-moi savoir 

Merci,

JeanOlivier, 2024-11-11:
Je vais essayer de faire ça cette semaine, merci du support et de la patience!

Donc est-ce que le HP600ZB va être mieux intégré que ce que j'ai en ce moment via ce PR, ou il faut attendre le reste des tests via zha-toolkit avant ça?

claudegel, 2024-11-12:
Pour l'instant j'ai migré la branche HP6000ZB dans le master et j'ai fais un PR pour l'ajouter dans ZHA. Avec le scan je pense qu'on va trouver de nouveaux attribut car il y a deux endpoint, 1 et 2 pour cet appareil et il peut gérer le chauffage et la climatisation ainsi que la ventilation. Donc il y a certainement plusieurs attributs ajoutés dans le cluster 0xff01 et possiblement dans dautres cluster.

claudegel, 2024-11-25:
le PR a été inclu dans zha-device-handler et va apparaitre lors de la prochaine mise a jour de HA/ZHA. Ça serait bien si vous pouviez faire un scan avec ZHA-toolkit.scan_device. Puisque les HP6000ZB-xx ont deux endpoints, 1 et 2 il y a certainement des attributs dans le manufacturer cluster qui nous manque. Entre autre pour les swing mode vertical et horizontal.

JeanOlivier, 2024-11-27:
Bon j'ai réussi à prendre le temps de faire les patches et lancer le scan. 
Dites-moi si tout est beau ou si vous avez besoin d'autre choses.
C'est un gros data dump, donc cherchez "HP6000ZB-MA" pour passer au second endpoint.

File: HP6000ZB-MA_Sinope-Technologies_XXXXXXXXXX_01_scan_results.txt
``` json
{
  "ieee": "XX:XX:XX:XX:XX:XX:XX:XX",
  "nwk": "0x34ed",
  "model": "HP6000ZB-MA",
  "manufacturer": "Sinope Technologies",
  "manufacturer_id": "0x4508",
  "endpoints": [
    {
      "id": 1,
      "device_type": "0x0307",
      "profile": "0x0104",
      "in_clusters": {
        "0x0000": {
          "cluster_id": "0x0000",
          "title": "Basic",
          "name": "basic",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "zcl_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 8
            },
            "0x0001": {
              "attribute_id": "0x0001",
              "attribute_name": "app_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 33
            },
            "0x0002": {
              "attribute_id": "0x0002",
              "attribute_name": "stack_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 0
            },
            "0x0003": {
              "attribute_id": "0x0003",
              "attribute_name": "hw_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 1
[... 3177 lines cut ...]
          },
          "commands_received": {},
          "commands_generated": {
            "0x01": {
              "command_id": "0x01",
              "command_name": "1",
              "command_args": "not_in_zcl"
            },
            "0x03": {
              "command_id": "0x03",
              "command_name": "3",
              "command_args": "not_in_zcl"
            },
            "0x06": {
              "command_id": "0x06",
              "command_name": "6",
              "command_args": "not_in_zcl"
            }
          }
        }
      }
    }
  ]
}
```

claudegel, 2024-11-27:
C'est parfait, j'ai les deux clusters avec plein de nouveaux attributs. je suppose que ce sont les controles du swing verticale et horizontale. je vais commencer par les ajouter dans thermostat.py pour que leur valeurs soient accessibles et modifiables via ZHA. Ensuite on va tenter de nommer les attributs du cluster 0xff01 qui est spécifique à Sinopé pour cet appareil.
Je vous avise quand les modifications seront prêt

claudegel, 2024-11-28:
Bon j'ai ajouté tous les attributs qui sont specifiques pour le HP6000ZB-MA. Je pense que ce seront les même que pour les deux autres HP6000ZB-ge et HS.
J'ai créer une nouvelle branche HP6000ZB-MA
https://github.com/claudegel/sinope-zha/tree/HP6000ZB-MA
Seul thermostat.py a changé. mais tu dois avoir la derniere version de sinope-zha, v0.1.9 sinon ramasse tous les fichiers.
Je suppose que le endpoint 1 est pour le mode chauffage et le 2 pour la climatisation. Faudra voir.
En redémarrant tu devrais voir dans ZHA les nouveaux attribut sur le cluster 0xff01, SinopeTechnologiesManufacturerCluster et dans SinopeTechnologiesThermostatCluster. Dans ce dernier, l'attribut 0x0437 qui vaut 100 je pense que c'est le target_temperature_step. 100 ça veut dire 1 oC et sur la carte de thermostat pour ma thermopompe ça augmente par incrément de 1 degré. Donc possiblement que cet attribut fixe cette valeur. On pourrait essayer de la changer pour 200, 2oC pour voir si sur la carte du HP6000 le setpoint va changer par 2oC.
Bref il faut jouer avec les valeurs des attributs pour trouver ce a quoi ils servent. Pour les changer tu vas dans ZHA, sélectionne le HP6000ZB-MA et clic sur les trois petits point verticaux pour selectionner l'option gérer l'appareil zigbee.
[image]
Dans la page qui ouvre tu choisis le cluster sur la première ligne et l'attribut sur la deuxième ligne. Au bas il y a deux boutons, lire et écrire. Tu peux lire la valeur de l'attribut et la changer en écrivant une nouvelle valeur. Dans a doc, readme.md, de sinopé-zha pour la branche HP6000ZB-MA j'ai mis tous les attributs avec les valeur trouvées dans ton scan.

Dernière chose. Quand on ajoute les HP6000ZZB-xx dans Neviweb il faut sélectionner le model de thermopompe et Sinopé envoi les capacités de swing vertical, horizontal et de fan spécifique à cet appareil. Je sais pas si tu as un GT130 pour mettre to HP6000ZB-MA dans neviweb et sélectionner ta thermopompe pour ensuite le remettre dans ZHA pour voir si des valeurs vont changer dans les cluster. Ça pourrait aider.

claudegel, 2024-11-28:
As-tu un printscreen de la carte de ton HP6000ZB-MA dans ZHA comme ci haut pour voir ce qui est affiché.

JeanOlivier, 2025-02-16:
> As-tu un printscreen de la carte de ton HP6000ZB-MA dans ZHA comme ci haut pour voir ce qui est affiché.

Est-ce que ceci est suffisant? 

[image]
[image]
[image]
[image]
[image]

Désolé j'ai vraiment manqué cette notification. Vaut mieux tard que jamais.

claudegel, 2025-03-21:
Merci, je vois que les controle, capteurs etc sont doublés. C'est qu'il y a deux endpoint sur l'appareil ce qui double tous les clusters. ZHA lit les deux et génère les sensor et autres contrôles. Je pense qu'il y en a un pour le chauffage et un pour la clim mais ZHA fait pas la difference.
Ca aide a comprendre le fonctionnement. Merci,

claudegel, 2025-04-25:
C'est surprenant car dans ZHA les ieee des appareils sont leur unique_id. Les ieee ne changent pas et donc mais si HA redémarre il va voir le même ieee et donner le même nom à l'appareil.
Peux-tu me dire c'etait quoi les patch de zha toolkit. Ca fait un bout.

claudegel, 2025-04-25:
On est présentement a transférer les quirks vers ZHA V2. je les utilises sur mon HA. Il me reste juste light.py a transférer, j'ai encore quelques erreur pour les lumières mais les autres fichier fonctionnent bien. Pas d'erreur pour thermostat.py, switch.py et sensor.py.
Si tu veux tu peux copier thermostat.py de la branche si dessous:
https://github.com/claudegel/sinope-zha/tree/quirks-V2
et remplacer thermostat.py dans config/zhaquirks. la détection des appareils est differente. Possiblement que ca va corriger ton bug.
Mais j'aimerais bien savoir si tes deux endpoint ont le même ieee. Si c'est pas le cas le bug viens de là.

claudegel, 2026-06-13:
Salut,

Depuis 2025 on a beaucoup progressé dans sinope-zha. Tout est maintenant en version ZHA-V2. Este-ce que vous avez installé seulement ZHA ? ou vous avez installé mes quirks sinope-zha dans HA. Il y a encore un petit bug à trouver où certain appareils Sinopé Zigbee  sont mal détectés si on les installe directement dans ZHA avec les quirks sinope-zha installé. Peut-être vous pourriez m'aider à corriger ce bug.
En premier dites-moi comment vous êtres installé avec ZHA. Seulement le ZHA de base avec HA ou vous avez installé mes quirks.