claudegel/sinope-zha #44: Valve VA4220ZB - Questions

maxperron, 2024-05-27:
Bonjour,

J'aurais quelques questions concernant l'intégration ZHA de la valve VA4220ZB.

Premièrement, merci énormément pour le travail sur tous les équipements Sinope!

Note: J'ai ajouté la valve avec ZHA sans utiliser le zhaquirks de ce répertoire.

[image]

J'avais des questions sur les entités créées:
1. L'entité "Ouverture" semble être toujours à "Closed" peut importe l'état de la valve, que représente-t-il?
2. Que représente "Zone IAS" ?
3. Est-ce qu'il y a une façon d'avoir l'information sur la batterie?
4. Le device type de l'entité "Summation delivered" est "Water Metering" ce qui fait que l'ont ne peut pas l'utiliser dans la dashboard Energy, puisque qu'il doit être de type "water" seulement. Je ne sais pas si c'est possible de le changer dans une version future.
[image]
Pour le moment, j'ai créé un template avec le bon device type:
[image]

Merci!

claudegel, 2024-05-28:
Je vous reviens demain. Pouvez-vous publier la signature de votre valve dans ZHA.
parametres, appareils et services, zha, sélectionnez votre valve et cliquez sur les trois petits points verticaux a coté de reconfigurer. Dans la fenêtre qui s'ouvre cliquez sur gérer l'appareil zigbee puis choisir le 3e onglet.

claudegel, 2024-05-29:
Vous avez la même valve que moi sauf que la mienne est au firmware 010402. je vais la remettre dans neviweb pour faire la mise a jour.
Les sensors qui apparaissent ne sont pas produits par les quirks mais directement par ZHA. Je pense qu'ils créent les sensors sur les cluster/attributs qui font du reporting.
Maintenant que les quirks sont pas mal matures je vais regarder pour ajouter les sensors qui manquent. Par contre avec ZHA tool-kit on peut créer les sensor qu'on a besoin. Par exemple pour le voltage et le % de la batterie c'est le cluster  0x0001, PowerConfiguration, attribut 0x0020 et 0x0021 il y a un service de ZHA tool-kit qui permet de créer un sensor pour suivre l'état des ces attributs.
Dans votre carte il y a commutateur (hidden) je n'ai pas ca sur la mienne est-ce quelques chose que vous avez ajouté ?
Est-ce possible d'exécuter zha_toolkit.scan_device sur votre valve.
```
service: zha_toolkit.scan_device
data:
  ieee: 50:0b:91:40:00:02:26:6d <-ieee of your valve
  endpoint: 1
  event_success: my_read_success_trigger_event
  event_fail: my_read_fail_trigger_event
  event_done: my_read_done_trigger_event
```
Ca va créer un fichier dans config/scans/ qui dompe tous les clusters et attributs de votre valve. ceci me permettrait de comparer et voir s'il n'y a pas de nouveaux attributs pour cette valve.
Concernant vos questions. L'entité ouverture sur ma valve est aussi toujours à closed je vais essayer de trouver quel attribut elle suit. Pour la zome IAS c'est l'alarme en cas de fuite détectée par le débit mètre. Pour la batterie, je l'ai indiqué plus haut, je vais retrouver le service qui permet de créer le sensor pour la  batterie. Pour le summation delivered c'est effectivement le débit mesuré en L/hr. je vais regarder pour le modifier mais en attendant pour vouvez créer un sensor de type water qui va suivre le summation delivered et vous permettre de l'inclure dans le module énergie.

claudegel, 2024-05-29:
le service suivant permet de récupérer la valeur de la batterie
```
service: zha_toolkit.execute
data:
  command: attr_read
  ieee: 50:0b:91:40:00:03:ed:b0
  cluster: 0x0001
  attribute: 0x0021
  state_id: sensor.bat_percent
  allow_create: True
  tries: 100
```
Ca va créer un sensor.bat_percent et y inscrire la valeur de battery percent remaining. Il suffit de faire une automation qui va exécuter ce service selon la fréquence requise.

maxperron, 2024-05-29:
Merci pour l'information, je vais créer les sensors avec  ZHA tool-kit.

> Dans votre carte il y a commutateur (hidden) je n'ai pas ca sur la mienne est-ce quelques chose que vous avez ajouté ?

Oui, nous pouvons changer le type d'une switch dans Home Assistant maintenant, quand je l'ai changé pour Valve, HA a désactivé la switch.
[image]

> Est-ce possible d'exécuter zha_toolkit.scan_device sur votre valve.

J'ai attaché le fichier généré.
[VA4220ZB_Sinope-Technologies_400005b318_01_scan_results.txt](https://github.com/claudegel/sinope-zha/files/15490744/VA4220ZB_Sinope-Technologies_400005b318_01_scan_results.txt)

claudegel, 2024-05-30:
Merci pour le fichier. Je vais travailler pour ajouter directement le type valve dans ZHA pour sortir les valves du groupe switch comme je l'ai fais pour neviweb130 ou le type valve suit HA valve. Dans HA le type valve est nouveau de cet année. 
Je vais aussi regarder pour créer directement les sensors qui nous intéressent. Ca va demander un peu de temps mais c'est la suite logique aux quirks.

krokets, 2024-06-04:
Bonjour,

Je suis aussi sur le cas de "Zone IAS".  J'ai remarqué que IAS devient vrai  au moment que la valve détecte un débit anormal et aussitôt que la valve se ferme, il retombe à faux. De mon côté, j'aimerais seulement être averti d'un débit anormal sans fermeture de valve, j'ai donc mis 0x0001 dans FF01h:0253h pour alerte seulement. Par contre, Zone IAS ne devient plus jamais vrai.  J'ai pensé que FF01h:0230h alarm_flow_threshold devrait être à ON, mais sans succès. 

Autre point étrange, en vérifiant les différents attributs, je tombe sur FF01h:0231h alarm_options qui est réglé à _AlarmAction.undefined_0x04_

claudegel, 2024-06-04:
J'ai observé aussi cette valeur pour AlarmAction. je pense que c'est lorsque le flowmeter n'est pas définit. Je l'ai ajouté pour la prochaine mise a jour. je pense qu'une fois qu'on a configurer le flowmeter il faut chager la valeur de AlarmAction pour une des autres valeur ci-dessous.
        ```
Nothing = 0x00
Notify = 0x01
Close = 0x02
Close_notify = 0x03
No_flow = 0x04 
```
Ma valve est dans Neviweb pour un update de firmware mais je vais la ramener dans ZHA. Mais avant je vais sniffer la ligne pour voir quel attributs sont modifiés quand on change les valeurs des options dans neviweb.