claudegel/sinope-zha #45: RM3250ZB, quirks ne s'applique pas

krokets, 2024-06-04:
Bonjour

J'ai un RM3250ZB et j'ai remarqué que le quirks n'est pas appliqué, car il y a le cluster 0x0002 dans la signature comparativement au code dans switch.py.

Voici la signature:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4508, maximum_buffer_size=71, maximum_incoming_transfer_size=43, server_mask=10752, maximum_outgoing_transfer_size=43, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0002",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0x0b05",
        "0xff01"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Sinope Technologies",
  "model": "RM3250ZB",
  "class": "zigpy.device.Device"
}
```

J'ai fait la modification local pour ajouter le cluster `DeviceTemperature`. Avec ce changement, le quirks est appliqué, mais la température ne se met pas à jour. 

```python
class SinopeTechnologiesLoadController(CustomDevice):
    """SinopeTechnologiesLoadController custom device."""

    signature = {
        # <SimpleDescriptor(endpoint=1, profile=260,
        # device_type=2, device_version=0,
        # input_clusters=[0, 2, 3, 4, 5, 6, 1794, 2820, 2821, 65281]
        # output_clusters=[3, 4, 25]>
        MODELS_INFO: [
            (SINOPE, "RM3250ZB"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha_p.PROFILE_ID,
                DEVICE_TYPE: zha_p.DeviceType.ON_OFF_OUTPUT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    Diagnostic.cluster_id,
                    SINOPE_MANUFACTURER_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    CustomDeviceTemperatureCluster,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    Diagnostic.cluster_id,
                    SinopeManufacturerCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

[image]

claudegel, 2024-06-05:
Le RM3250ZB c'est pas un calypso. La température est celle interne de la switch pour savoir si elle chauffe trop.
J'ai un vieux RM3250ZB qui est détecté parfaitement sans le cluster 0x0002.
Donc on peut pas modifier le quirk existant mais en ajouter un nouveau. Est-ce que ton appareil est récent? et est-ce que sur neviweb ça mentionne un model G2 comme pour les thermostat et les interrupteurs ?
Pourrais-tu installer zha toolkit et faire un domp des clusters ca va me permettre de valider quel attribut on doit cibler.
Finalement quel est le firmware de ton appareil?

claudegel, 2024-06-05:
Ok j'ai fais la modif dans une nouvelle branche:
https://github.com/claudegel/sinope-zha/tree/new-rm3250
il y a juste le fichier switch.py qui change.
Ton appareil sera détecté comme switch.SinopeTechnologiesNewLoadController

claudegel, 2024-06-05:
Reste a valider les question que je t'ai posé ci-haut pour voir si on ajoute des attributs dans le cluster 0x0002 ou le 0xff01

krokets, 2024-06-05:
Voici les détails dans neviweb. J'ai acheté il y a 1 ans et demi à mon souvenir. 

[image]

J'ai remarqué une différence entre la température avant l'application du quirks et après. Je vois une différence de 2degC.

[image]

J'ai l'impression que la division par 100 par la class DeviceTemperature(Sensor) et la multiplication par 100 dans la class CustomDeviceTemperatureCluster(CustomCluster, DeviceTemperature) fait perdre beaucoup de précision 

[RM3250ZB_Sinope-Technologies_400001f98a_01_scan_results.txt](https://github.com/user-attachments/files/15585734/RM3250ZB_Sinope-Technologies_400001f98a_01_scan_results.txt)

claudegel, 2024-06-07:
Les appareils zigbee enregistrent la température au format de oC*100. Donc 29 oC est enregistré comme 2900. ZHA transforme alors toutes les valeurs de température en divisant par 100. Le probleme de l'attribut deviceTemperature est que Sinopé l'enregistre  comme 29oC et donc ZHA le transformait en 0.29oC.
Le quirks vient corriger en multipliant la valeur de l'attribut par 100 avant que ZHA la divise par 100.
Mon sinope-zha est une copie des quirks dans ZHA qui permet de développer sans briser ZHA. Quand ma nouvelle version fonctionne je pousse un PR dans ZHA.
Présentement et ZHA et sinope-ZHA font la même correction en multipliant la valeur de l'attribut par 100.
Dans les deux cas les températures affichées de 29 et 31 ont subies les même correction via ZHA ou sinope-ZHA. Il n'y a pas d'arrondissement.
Si la modification que j'ai fais dans switch.py fonctionne votre RM3250ZB sera détecté comme
switch.SinopeTechnologiesNewLoadController
Si tout est beau je vais merger cette branche et faire un PR sur ZHA

krokets, 2024-06-07:
Pour la détection, c'est réglé. Pour ce qui est du refresh rate de l'attribute, il se fait à différence de plusieurs heures. Je présume que c'est le module qui ne met pas jour rapidement ?

claudegel, 2024-06-07:
Pour les appareils zigbee, celles-ci font du reporting. Elle publient leur differents attribut selon un horaire défini ou si la valeur change. Pour un thermostat, la temperature de la pièce varie beaucoup alors ça va publier souvent et puisque la valeur est importante le thermostat va publier selon une fréquence plus rapide.
Pour une température interne de l'appareille qui change peu, il est probable que Sinopé a mit une fréquence basse. Donc si la temperature de l'appareil est stable elle ne publie pas souvent.
Je pense qu'il est possible de modifier la fréquence du reporting. Dans mon readme.mc j'explique comment on fait pour le reporting de d'autres attributs.

claudegel, 2024-06-09:
Donc si la détection de votre RM3250ZB fonctionne je vais publier vers ZHA pour ajouter au quirks.

krokets, 2024-06-09:
Bonjour 

Désolé du retard. Oui il fonctionne! Merci pour l'aide

claudegel, 2024-07-20:
Ok parfait j'ai publié et c'est maintenant dans ZHA.
Laissez-moi savoir s'il y a autre chose

closed 2024-07-20 by claudegel