claudegel/sinope-zha #48: Sedna Zigbee Valve Switch Gone After 2024.8.0 Update

mtwo, 2024-08-07:
The switch control on my Sedna VA4220ZB disappeared after updating Home Assistant to 2024.8.0. I've attempted to re-pair it both with and without the custom quirk, and in both cases I no longer have a way of opening or closing it. I _think_ that I read something in a previous release about valves no longer needing to be represented as switches, could this be related?

saucepwn, 2024-08-07:
This is with the ZHA integration

claudegel, 2024-08-08:
Hi, I just made the update to 2024-8.0. I'll check.

Yes the valve entity is now separated from the switch but ZHA didn't make the change until now. It is possible that in 2024.8.0 they finally added the valve. If yes I'll  push a PR to my sinope-zha and when stable to ZHA quirks.
Let me check if the valve entity are now in ZHA.
You can add logging in HA for ZHA. Add this to configuration.yaml and restart HA
```
logger:
  default: warning
  logs:
    homeassistant.custom_components: debug
    homeassistant.components.zha: debug
    zhaquirks: debug
    zigpy.zcl: debug
    zigpy.application: debug
```
You should get something in the home-assistant.log

claudegel, 2024-08-08:
I have a sedna VA4220ZB in Neviweb I'll switch it to my sinope-zha to be able to make the changes if valve entity were added to ZHA.

claudegel, 2024-08-08:
My valve is recognized as switch.SinopeTechnologiesValveG2 with sinope-zha and should be the same with zha quirks.
But there is only sensors and number. No button to open or close the valve.  Here is the signature:
```
"endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0003",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0402",
        "0x0404",
        "0x0500",
        "0x0702",
        "0x0b05",
        "0xff01"
      ],
      "output_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Sinope Technologies",
  "model": "VA4220ZB",
  "class": "switch.SinopeTechnologiesValveG2"
```
Do you have a flowmeter connected to your valve ? and is it a gen2

claudegel, 2024-08-08:
Do you have cluster 0x0404 in your signature ?
Also do you have zha-toolkit installed ? Could help to get a scan of all clusters.

mtwo, 2024-08-08:
I have a flow meter and I _think_ that it's a gen 2. Here's the signature, 0x404 is there:

```
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 1,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4508,
    "maximum_buffer_size": 71,
    "maximum_incoming_transfer_size": 43,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 43,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0003",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0402",
        "0x0404",
        "0x0500",
        "0x0702",
        "0x0b05",
        "0xff01"
      ],
      "output_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Sinope Technologies",
  "model": "VA4220ZB",
  "class": "zhaquirks.sinope.switch.SinopeTechnologiesValveG2"
}
```

I have zha-toolkit installed, though I think that the HA update also may have broken it, with the shift from services to actions. I can't call its methods through the developer tools anymore.

mtwo, 2024-08-08:
FYI it also looks like the flowmeter is reporting 0 after the update, though this could be due to me deleting and re-pairing the device without the quirk

claudegel, 2024-08-09:
Did you reconfigure the flowmeter. If not you will need zha toolkit but it is also broken sinse the update.

mtwo, 2024-08-09:
ZHA toolkit just got updated, doing that now

mtwo, 2024-08-09:
Hmm, ZHA toolkit may still be having issues even after the update

[image]

claudegel, 2024-08-09:
Will have to investigate on that. many bug with the last HA update, 2024.8.0. My thermostats on ZHA show time 4 hours ahead.

claudegel, 2024-08-10:
Ok spent couple hours with the ZHA developper and it urn out it was a bug in one of the library. it is fix and merged into zha and should appear probably in release 2024.8.2
Thans for reporting this

mtwo, 2024-08-10:
That’s great! Thank you so much for this and all of the work that you do!

mtwo, 2024-08-16:
Just updated to .2 and I can confirm that it's back!

closed 2024-08-16 by mtwo