claudegel/sinope-zha #49: Cannot get energy usage from SP2600ZB

houlejm, 2024-08-31:
If I try to add energy usage from SP2600ZB to homeassistant energy dashboard it does not work. From https://www.home-assistant.io/docs/energy/faq/#troubleshooting-missing-entities I see that the sensor is missing some attributes.

[image]

I can get around the problem with a template:
  - name: Énergie interrupteur
    unique_id: energie_interrupteur_1
    state: "{{ states('sensor.interrupteur_prise_1_summation_delivered') }}"
    state_class: total_increasing
    device_class: energy
    unit_of_measurement: "kWh"

This new sensor works.

[image]

Thanks

claudegel, 2024-08-31:
it is just because in your zha sensor sensor.interrupteur_prise_1_summation_delivered there is no unique_id attribute
without unique_id the energy dashboard do not monitor it.
```
state_class: total_increasing
zcl_unit_of_measurement: 0
status: NO_ALARMS
device_type: Electric Metering
unit_of_measurement: kWh
device_class: energy
friendly_name: Prise video Summation delivered
```
We are working on code remanufacturing in ZHA and I'll check to see if I can add unique_id attribute. Until that you will need to create specific sensor with unique_id to monitor your energy

houlejm, 2024-09-01:
I do not believe that the unique_id is the reason.
I tested that by removing the unique in the template that works and it still works.

 - sensor:
   - name: Énergie interrupteur
      state: "{{ states('sensor.interrupteur_prise_1_summation_delivered') }}"
      state_class: total_increasing
      device_class: energy
      unit_of_measurement: "kWh"

It is more the device attributes that are missing:
  state_class: total_increasing
  device_class: energy
  unit_of_measurement: "kWh"
as documented in: https://www.home-assistant.io/docs/energy/faq/#troubleshooting-missing-entities

I have a  RM3500ZB  that works ok without using a template and that SP2600ZB used to work in the energy dashboard.

Anyway, I am OK for now. Keep the goog work with ZHA and Sinope

claudegel, 2024-09-01:
What is the quirk name for your device SP2600ZB, Quirk: switch.SinopeTechnologiesSwitch ? or Quirk: switch.SinopeTechnologiesSwitch_V2
Can you make a device scan with ZHA 🧰 Toolkit: Scan Device and send me the file it will create in config/scans

houlejm, 2024-09-01:
Tel que demandé:

[SP2600ZB_Sinope-Technologies_0018471abf_scan_results.txt](https://github.com/user-attachments/files/16830796/SP2600ZB_Sinope-Technologies_0018471abf_scan_results.txt)

JM

claudegel, 2024-09-03:
No I already have many SP2600ZB. I'll check if we can add sensor for summation_delivered with all the attributes it need to be usable by the energy daskboard directly in ZHA. 
As there is a lot of code changed in ZHA now it could take a little time. You can have a look at zha-toolkit.  With it you can create a sensor from any cluster/attribute and it will update the data automatically.

arnacb, 2025-04-26:
Hi,

I just purchased an SP2600ZB, and the Summation Delivered feature seems to be broken.

I'm not a programmer, and I've only been working on this for a few days, does anyone have any idea what could be wrong?

Thanks in advance!

claudegel, 2025-04-26:
ZHA is changing lot of thing now. The most important is that we are porting all quirks to quirks_V2. All reporting attributes are set as separate sensor that can be seen in the device card.

Current summation delivered is set as a separate sensor with a name like this 
sensor.your_device_summation_delivered
or in french sensor.your_device_consommation
See my card for a SP2600ZB firmware 0.1.8. You should have the same

Are you using my sinope-zha quirks or directly the zha quirks ?

[image]

claudegel, 2025-04-26:
If you are with my sinope-zha you can test with the V2 quirks.
Go in branch:
https://github.com/claudegel/sinope-zha/tree/quirks-V2
Copy the file switch.py and replace the one you have in config/zhaquirks then restart HA. On first restart you will probably have to delete sensor in the card like the energie on the picture. Then restart again and your all set. you will have access to the summation sensor directly.

[image]

arnacb, 2025-04-26:
Hi 

thanks for the quick answer here hat i get with your Quirk :

[image]

i ask Chat GPT and it propose to add this in the function but it cause the sensor to stop updating.

SinopeTechnologiesMeteringCluster
        # Attribut propriétaire réel utilisé par l’appareil
        current_summation_delivered: Final = foundation.ZCLAttributeDef(
            id=0x0090, type=t.uint32_t, access="rp", is_manufacturer_specific=True
        )

        # Attribut standard attendu par ZHA/Home Assistant
        current_summation_delivered_std: Final = foundation.ZCLAttributeDef(
            id=0x0000, type=t.uint64_t, access="rp"
        )

claudegel, 2025-04-26:
Attribute for current_summ_delivered is already in the cluster 0x0702, metering cluster.
I've altered this cluster to add description of the values of two specific attributes 0x0200, status and 0x0300, unit_of_measure, to reflect what values Sinopé is using for those attributes. Then I rename it to SinopeTechnologiesMeteringCluster.
Attribute 0x0090 just don't exist in cluster metering and the SP2600ZB firmware don't have attribute 0x0090 in his metering cluster. Only attribute 0x0000 which is the one that we are reading in the quirks. We don't need to add attribute 0x0000 because SinopeTechnologiesMeteringCluster is a subclass of the metering cluster and already contain the 0x0000 attribute.
If you want to find which cluster/attributes are present in your SP2600ZB then you can use zha toolkit custom_component to scan the device and get a complete map of the clusters and attributes. You can install zha toolkit via HACS. Then I'll tell you the command to execute to get the complete cluster map for your device. It could be interesting to see if your SP2600ZB have a different clusters/attributes list. If yes we can make an update for the quirks. 

What I see on your picture, the summation delivered show 3.006 kwh which should be correct for a newly installed device. I don't know what is connected on your device. 
In the quirks we have a problem with the value of summation delivered because old SP2600ZB don't have the divider 1000 and the new devices have it.

So first I think we can start by scanning your device to see if it has different attributes.

claudegel, 2025-04-26:
By the way you can see what is the value in cluster 0x0702, attribute 0x0000 by clicking on the three vertical dot aside of the RECONFIGURE in blue. You can see it in the picture you've post.
When you clic, it open a popop. chose manage zigbee device. In the window that open choose the cluster on the fisrt line, SinopeTechnologiesMeteringCluster. Then below ATTRIBUTS chose which attribute you want to see. In the list you will find current_summ_delivered. After just clic on read attribute at the bottom. You should see your value 3006 which is in watt.