claudegel/sinope-zha #50: MC3100ZB (MultiController) wrong internal temperature readings

bassdr, 2024-09-30:
Hi, first of all, thank you very much @claudegel for all the work you've done to integrate sinope's devices to HA. I use your work a lot as I prefer encouraging a local brand and I am an opensource enthousiast.

I am posting this here but I am not yet sure if this is a hardware problem with my device, I am starting to investigate the issue here but I could also contact Sinope as the temperature reading never worked (the device is quite new).

So, the problem I have is that the temperature reading of my device is always a bit off. Actually about 1 degree lower in the 20-25 C range, it gets worse when then temperatures are lower, and I can notice no difference at around 28C.

At first sight, it looks like the internal sensor is mis-calibrated, or just bad. I started looking into the quirks here to see if there would be a way to calibrate the internal sensor. In the meantime, I conntected a 10K sensor to the "external" sensor (it's actually inside the device anyway), as this one seem acurate.

Would you have any clue on what could be wrong? BTW I also connected the device to neviweb GT130 and it does the same, tho I connected it to ZHA first. It's totally possible it's a hardware issue like I said.

Here's the comparison I did last night: You see internal vs. external 10k sensort vs. 3rd party temperature sensor with a very bad refresh rate.
[image]

claudegel, 2024-09-30:
Sorry your device is a MC3100ZB or a Thermostat?

bassdr, 2024-09-30:
mC3100ZB

claudegel, 2024-10-01:
TThe internal sensor is for the device temperature, not for room temperature. This is why there is aanother connection for an external sensor. I'm using this one with a floor sensor which is installed ooutside to get outside temperature and send that back to all my thermostats. I've talked to Sinopé wwhen I got mine because I noticed that there was a slight difference with the room temperature and tthey told me that it was the device temperature and not the room temperature.

closed 2024-10-01 by bassdr