claudegel/sinope-zha #56: SP2610ZB state cycle a

krokets, 2024-12-14:
Bonjour Claude

J'ai une prise SP2610ZB qui ferme et ouvre automatiquement sans intervention ou automation pendant 1 seconde environ. J'ai remarqué que ce problème arrive en même temps que le logbook affiche "switch_livingroom Summation delivered changed to 1775.199". Est-ce que tu as déjà expérimenté ce problème? C'est le seul périphérique qui affiche sa consommation dans le logbook. 

Voici un exemple du logbook: 

[image]

claudegel, 2024-12-15:
J'ai eu une prise SP2610Zb qui faisait la même chose. Elle était installée au plafond pour alimenter un cinema maison. A tout moment elle fermait pour quelques second et se rallumait sans raison. Pas d'automation. Je l'ai enlevée du plafond et placé au mur pour un autre usage moins critique, en position verticale. Elle fonctionne bien . J'en ai déduit que le fait qu'elle soit placée au plafond, à l'horizontale, causait une surchauffe dans un composant et ça coupait le courant pour quelques secondes. J'en ai parlé à Sinopé et leur ai retournée la switch mais celle qu'ils m'ont envoyé avait le même problème lorsqu'installée au plafond. Installée au mur, même avec des appareils énergivores comme un congélateur ne cause aucun problème. Pas de coupure de courant.
Est-ce que la votre est au plafond ?

claudegel, 2024-12-15:
Ce produit est maintenant discontinué chez Sinopé. Ils ont probablement trouvé un bug et la conception était pas terrible avec la prise contrôlée en haut. Lorsqu'on branche un appareil dont le connecteur est a 90 deg, congélateur, ça bloque la prise inférieure.

krokets, 2024-12-15:
Bonjour Claude,

Merci de ta réponse. Ça semble être le même comportement. Chez moi, la prise est sur un mur et elle sert pour la lumière du salon, donc elle est très peu sollicitée. À mon souvenir, avec le gt130, je n'avais pas ce problème. Je vais réessayer à nouveau. Par contre, sais-tu pourquoi la consommation s'affiche dans le logbook ? Il semble avoir un bug ? C'est peut être relié?

claudegel, 2024-12-16:
J'avais le même problème dans Neviweb et ZHA par contre j'ai présentement 6 SP2610ZB et il y a rien dans les log pour Summation delivered.
Est-ce que ton appareil est ajouter au panneau energie? Si le module énergie suit la consommation il est possible que le fait d'interroger l'appareil toutes les heures pour avoir le Summation delivered génère une ligne dans les log.

krokets, 2025-01-25:
Bonjour Claude 

Simplement pour mettre à jour l'issue, j'ai enlevé et remis dans home assistant et depuis, je n'ai plus de problème. Par contre, l'entité summation delivered n'est plus disponible pour ajouter dans le panneau énergie, mais c'est un demi mal. Merci pour ton support.

closed 2025-01-25 by krokets

houlejm, 2025-01-25:
@krokets 
Bonjour.
J'avais un problème semblable avec un SP2600zb. Je ne pouvais ajouter la consommation dans le tableau de bord énergie. Il manquait des attributs dans le sensor summation_delivered.
Le problème est réglé maintenant, mais je l'avais contourné en ajoutant ceci dans configuration.yaml
`homeassistant:
    customize:
    sensor.interrupteur_prise_1_summation_delivered: 
      state_class: total_increasing
      device_class: energy
      unit_of_measurement: "kWh"
`
Mets ton sensor à la place de: "sensor.interrupteur_prise_1_summation_delivered"

claudegel, 2025-01-26:
Il y a eu beaucoup de changement dans ZHA dernièrement et je pense que les sensor des switch ne sont pas bien initialisé. C'est pas dans les quirks mais directement dans ZHA. Je vais regarder si je peut corriger lesetup.

krokets, 2025-01-28:
Merci @houlejm ça fonctionne. 

@claudegel Suite au fix de @houlejm on peut voir que l'unité de mesure était vide. Je ne sais pas si c'est dans le quirks qu'on doit peut-être indiquer cette valeur.

[image]

claudegel, 2025-01-28:
C'est dans ZHA que le sensor est créé. Je vais regarder ça car je suis a implémenter la création automatique des sensor dans neviweb130 et c'est la même approche.
Mais je vais ajouter une note dans la doc pour ça

claudegel, 2025-01-28:
J'ai vérifié sur un de mes SP2600ZB et j'ai ceci:
sensor.sinope_technologies_sp2600zb_ef246615_smartenergy_metering_summation_delivered
state_class: total_increasing
device_type: Electric Metering
status: NO_ALARMS
zcl_unit_of_measurement: 0
unit_of_measurement: kWh
device_class: energy
friendly_name: Prise video Consommation

sensor.sinope_technologies_sp2610zb_83356615_smartenergy_metering_summation_delivered
state_class: total_increasing
device_type: Electric Metering
status: NO_ALARMS
zcl_unit_of_measurement: 0
unit_of_measurement: kWh
device_class: energy
friendly_name: Prise télé salon Consommation

quel est le firmware # de vos SP26xx.

houlejm, 2025-01-28:
Mon SP2600zb est à 0.1.8 (vu par la GT130)

claudegel, 2025-01-28:
Les miens dans ZHA sont à 0.0.9
La nouvelle version dans Neviweb est a 0.1.8 C'est peut-être avec ceux-ci qu'il y a un bug. Je vais en mettre un dans ZHA pour voir

krokets, 2025-02-02:
J'ai 0.0.8 dans Neviweb avec une version matérielle 1

claudegel, 2025-02-03:
0.0.8 c'est pour le SP2610ZB qui a été discontinué. Pour le SP2600ZB il y a deux versions 0.0.9 et 0.1.8