claudegel/sinope-zha #60: Automation external temp

kirax999, 2025-01-13:
Hi, I have a problem with the automation to display the outside temperature. I'll send the code directly, it will be simpler.

[image]

do you have any idea?

claudegel, 2025-01-13:
Look like you have an ieee that is invalid. data[0] is your ieee: "{{ thermostats[repeat.index-1] }}"

Also add an id: at the first line
Mine is like this:
```
- id: Heat-OutdoorTemp
  alias: Heat-OutdoorTemp
  trigger:
    - platform: state
      entity_id: sensor.varennes_local
  variables:
    thermostats:
      - 50:0b:91:40:00:02:26:6d
      - 38:5c:fb:ff:fe:d9:ea:f4
  actions:
    - repeat:
        count: "{{thermostats|length}}"
        sequence:
          - action: zha.set_zigbee_cluster_attribute
            data:
              ieee: "{{ thermostats[repeat.index-1] }}"
              endpoint_id: 1
              cluster_id: 0xff01
              cluster_type: in
              attribute: 0x0010
              value: "{{ ( trigger.to_state.state|float * 100 ) |int }}"
  mode: single
```

claudegel, 2025-01-13:
Does sensor.local_temp is populated with temperature data ?

kirax999, 2025-01-14:
Je n'es pas encore relier mon capteur exterieur mais je pensais quil allais quand meme me laisser enregistrer la, pas dutout

[image]

Je ressayerais quand jaurais mit mon capteur exterieur je vous tiendrais au courant 

Y'a t'il un moyen de trouver facilement les cluster_id et autre info ?

claudegel, 2025-01-14:
Ben oui. Pour trouver tous les clusters et attributs zigbee, juste regarder le fichier readme.md de la page d'accueil de sinope-zha.

claudegel, 2025-01-14:
Pour la temperature esxterieure, en attendant ton capteur tu peux toujours installer un composant meteo de HA et utiliser son sensor de temperature. Quand le tiens sera en place tu pourra switcher.

syleishere, 2025-05-10:
Hi Claudegel, thank-you for your excellent work on Sinope devices, I contacted their support and they told me to order th1134zb-hc thermostat as that one does not have a mechanical relay so that is what I did(I'll need 3 more if this works out), it is now installed, love fact has color display, my first time setting up automations with File Editor. I did try with environment canada first, but icon code kept coming back "unknown" for whatever reason so I ended up ditching it and going with openweathermap as seems more reliable for weather code. Do these automations look correct for thermostat?

`- id: thermostats_outdoor_temp
  alias: Thermostats Outdoor Temperature
  trigger:
    - platform: state
      entity_id: 'sensor.openweathermap_temperature'
  variables:
    thermostats:
      - 94:34:69:ff:fe:b7:3a:0a
  actions:
    - repeat:
        count: "{{thermostats|length}}"
        sequence:
          - action: zha.set_zigbee_cluster_attribute
            data:
              ieee: "{{ thermostats[repeat.index-1] }}"
              endpoint_id: 1
              cluster_id: 0xff01
              cluster_type: in
              attribute: 0x0010
              value: "{{ ( trigger.to_state.state|float * 100 ) |int }}"
  mode: single
- id: thermostats_icons_daytime
  alias: Thermostats Icons Daytime
  triggers:
    - trigger: state
      entity_id: 'sensor.openweathermap_weather_code'
    - trigger: sun
      event: sunrise
      offset: "00:30:00"
  conditions:
    - condition: state
      entity_id: 'sun.sun'
      state: 'above_horizon'
  variables:
    weather_change:
      '200': Cloudrainsun2
      '201': Cloudlightning
      '202': Cloudlightning
      '211': Cloudrainsun2
      '212': Cloudrainsun2
      '300': Cloud3
      '301': Cloud3
      '311': Cloud3
      '500': Cloudrainsun
      '501': Cloudrainsun3
      '502': Cloudrainsun4
      '504': Cloudrain3
      '511': Cloudlightning2
      '520': Cloudrain
      '521': Cloudrain2
      '522': Cloudrain3
      '600': Cloudsnow2
      '601': Cloudsnow4
      '602': Cloudsnow3
      '613': Cloud
      '615': Cloud
      '616': Cloud
      '621': Cloudsnow3
      '622': Cloudsnow5
      '701': Cloudfog
      '711': Cloudfog
      '721': Cloudfog
      '731': Cloudfog
      '741': Cloudfog
      '751': Cloudfog
      '761': Cloudfog
      '800': Sun
      '801': Sun2
      '802': Cloudsun
      '803': Cloudsun2
      '804': Cloud2
    thermostats:
      - 94:34:69:ff:fe:b7:3a:0a
  actions:
    - repeat:
        count: '{{ thermostats|length }}'
        sequence:
          - service: zha.set_zigbee_cluster_attribute
            data:
              cluster_type: in
              ieee: '{{ thermostats[repeat.index-1] }}'
              endpoint_id: 1
              cluster_id: 0xff01
              attribute: 0x0013
              value: '{{ weather_change.get(states("sensor.openweathermap_weather_code"), ''Hide'') }}'
  mode: single
- id: thermostats_icons_nighttime
  alias: Thermostats Icons Nighttime
  triggers:
    - trigger: state
      entity_id: 'sensor.openweathermap_weather_code'
    - trigger: sun
      event: sunset
      offset: "00:30:00"
  conditions:
    - condition: state
      entity_id: 'sun.sun'
      state: 'below_horizon'
  variables:
    weather_change:
      "200": Cloudrainmoon2
      "201": Cloudrainmoon2
      "211": Cloudrainmoon2
      "212": Cloudrainmoon2
      "300": Cloud3
      "301": Cloud3
      "311": Cloud3
      "500": Cloudrainmoon
      "501": Cloudrainmoon4
      "502": Cloudrainmoon5
      "503": Cloudrainmoon5
      "504": Cloudrain3
      "511": Cloudrainmoon3
      "521": Cloudrain2
      "522": Cloudrain3
      "600": Cloudsnow6
      "601": Cloudsnow4
      "602": Cloudsnow3
      "613": Cloud
      "620": Cloudsnow5
      "621": Cloudsnow3
      "622": Cloudsnow5
      "701": Cloudfog
      "711": Cloudfog
      "721": Cloudfog
      "731": Cloudfog
      "741": Cloudfog
      "751": Cloudfog
      "761": Cloudfog
      "771": Cloudsnow5
      "800": Moonstar
      "801": Cloudmoon
      "802": Cloudmoon2
      "803": Cloudmoon3
      "804": Cloud2
    thermostats:
      - 94:34:69:ff:fe:b7:3a:0a
  actions:
    - repeat:
        count: '{{ thermostats|length }}'
        sequence:
          - service: zha.set_zigbee_cluster_attribute
            data:
              cluster_type: in
              ieee: '{{ thermostats[repeat.index-1] }}'
              endpoint_id: 1
              cluster_id: 0xff01
              attribute: 0x0013
              value: '{{ weather_change.get(states("sensor.openweathermap_weather_code"), ''Hide'') }}'
  mode: single
- id: thermostats_startup
  alias: Thermostats Startup
  trigger:
    - platform: homeassistant
      event: start
  variables:
    thermostats:
      - 94:34:69:ff:fe:b7:3a:0a
  action:
    - repeat:
        for_each: "{{ thermostats }}"
        sequence:
          - service: zha.set_zigbee_cluster_attribute
            data:
              ieee: "{{ repeat.item }}"
              endpoint_id: 1
              cluster_id: 0xff01
              cluster_type: in
              attribute: 0x0114
              value: 1 # Set time format 0=24h, 1=12h
          - service: zha.set_zigbee_cluster_attribute
            data:
              ieee: "{{ repeat.item }}"
              endpoint_id: 1
              cluster_id: 0xff01
              cluster_type: in
              attribute: 0x0136
              value: 0 # Disable weather icon timeout - DEFAULT 10800 sec (3hrs)
  mode: single`

syleishere, 2025-05-10:
I am having issue with icon disappearing after a period of time, still can't figure this out.

claudegel, 2025-05-10:
Sorry I was out. The weather icon that are sent to the thermostat are erased if there is no change or update for 3 hrs. This is the default. You can change timeout with the attribute weather_icons_timeout 0x0136. As you need to do it only once it is not necessary to make an automation you can change the value on this attributemanually.
Go to Parameters, ZHA, select your thermostat to view the card. Then clic on the three vertical dot at the wright of Reconfigure and choose manage zigbee device.
On the window that open, on the first line, cluster choose SinopeTechnologiesManufacturerCluster 0xff01. Then on the second line Attributes choose weather_icons_timeout 0x0136. It's the last one I think.
Then at the bottom of that window there are two buttons; read and write.
First read the attribute value. It's in seconds, probably 10800 which is the default, 3 hours. Then change the value for up to 18000 seconds (5 hours) and write the value. You can clic after on read to see thatthe value have been changed. Now you will have a timeout of 5 hours.
You can also add an automation that will send the active icons every 4 hours to reset the timeout counter.

claudegel, 2025-05-10:
For the timeout accepted values are 3600 to 18000

syleishere, 2025-05-11:
oww so that is the issue, I cannot use 0 there, I'd have to set to 18000, ok thank-you for that

syleishere, 2025-05-11:
Actually yeah only 5 hours is still a problem, I mean very possible forecast would not change from cloudy all day always still creating issues, so would need modded automation or new automation to update value. So this doesn't seem very efficient. One thing that should change frequently is temperature, so I guess the icon should just get updated along with that would be the best way.

claudegel, 2025-05-11:
If you want to test the new sinope-zha V2 you should be happy. All attributes that have some importance are now added directly as sensor. Automation do not need to use ieee anymore. for example the automation to send weather icons is a lot simpler:
for example you don't need to send the icon code to an attribute but instead you send it to a select entitie that represent that attribute and it take care of sending it to the thermostat. To change time from 24h to 12 h it is now a select menu in the device card. See screen capture below.
To test it just pick the 4 files from this link and replace the one you have in config/zhaquirks
https://github.com/claudegel/sinope-zha/tree/quirks-V2
```
- id: Icon-switch-night-v2
  alias: Icon-switch-night-v2
  initial_state: False
  description: "Weather icon changed when OpenWeatherMap weather code is updated during the night"
  mode: single

  triggers:
    - trigger: state
      entity_id: sensor.weather_code
    - trigger: sun
      event: sunset
      offset: "00:30:00"

  variables:
    weather_change:
      "200": Cloudrainmoon2
      "201": Cloudrainmoon2
      "211": Cloudrainmoon2
      "212": Cloudrainmoon2
      "300": Cloud3
      "301": Cloud3
      "311": Cloud3
      "500": Cloudrainmoon
      "501": Cloudrainmoon4
      "502": Cloudrainmoon5
      "503": Cloudrainmoon5
      "504": Cloudrain3
      "511": Cloudrainmoon3
      "521": Cloudrain2
      "522": Cloudrain3
      "600": Cloudsnow6
      "601": Cloudsnow4
      "602": Cloudsnow3
      "613": Cloud
      "620": Cloudsnow5
      "621": Cloudsnow3
      "622": Cloudsnow5
      "701": Cloudfog
      "711": Cloudfog
      "721": Cloudfog
      "731": Cloudfog
      "741": Cloudfog
      "751": Cloudfog
      "761": Cloudfog
      "771": Cloudsnow5
      "800": Moonstar
      "801": Cloudmoon
      "802": Cloudmoon2
      "803": Cloudmoon3
      "804": Cloud2

  conditions:
    - condition: state
      entity_id: sun.sun
      state: "below_horizon"

  actions:
    - action: select.select_option
      data:
        option: "{{ weather_change.get(states('sensor.weather_code'), 'Hide') }}"
        entity_id: select.salle_a_diner_weather_icons

- id: Icon-switch-day-v2
  alias: Icon-switch-day-v2
  initial_state: False
  description: "Weather icon changed when OpenWeatherMap weather code is updated during the day"
  mode: single

  triggers:
    - trigger: state
      entity_id: sensor.weather_code
    - trigger: sun
      event: sunrise
      offset: "00:30:00"

  variables:
    weather_change:
      "200": Cloudrainsun2
      "201": Cloudlightning
      "202": Cloudlightning
      "211": Cloudrainsun2
      "212": Cloudrainsun2
      "300": Cloud3
      "301": Cloud3
      "311": Cloud3
      "500": Cloudrainsun
      "501": Cloudrainsun3
      "502": Cloudrainsun4
      "504": Cloudrain3
      "511": Cloudlightning2
      "520": Cloudrain
      "521": Cloudrain2
      "522": Cloudrain3
      "600": Cloudsnow2
      "601": Cloudsnow4
      "602": Cloudsnow3
      "613": Cloud
      "615": Cloud
      "616": Cloud
      "621": Cloudsnow3
      "622": Cloudsnow5
      "701": Cloudfog
      "711": Cloudfog
      "721": Cloudfog
      "731": Cloudfog
      "741": Cloudfog
      "751": Cloudfog
      "761": Cloudfog
      "800": Sun
      "801": Sun2
      "802": Cloudsun
      "803": Cloudsun2
      "804": Cloud2

  conditions:
    - condition: state
      entity_id: sun.sun
      state: "above_horizon"

  actions:
    - action: select.select_option
      data:
        option: "{{ weather_change.get(states('sensor.weather_code'), 'Hide') }}"
        entity_id: select.salle_a_diner_weather_icons
```

[image]

syleishere, 2025-05-11:
Didn't even realize there was a version 2, thank-you for that, I noticed that did cause issues with blueprint that didn't work cause would not accept an ieee value.

There any way you can combine both daytime and nighttime icons for openweathermap examples to automatically use right one so we can combine updating icon along with outside temperature?

syleishere, 2025-05-11:
Maybe th1134zb-hc could have it's own class in thermostat.py with all those enums, and integration could be automatic like sending the time if they use openweathermap, since all those entities are pretty much the same for anyone who installed that, where environment canada will not work for everyone because I think it depends on weather station it picks with your lat/long weather it even supports sending that data at that weather station, plus it names all the entities based on that station. I think openweathermap is the solution here.

claudegel, 2025-05-11:
If you want to have your icon automation to be fired at each hour just add another trigger in your automation:
```
trigger:
    - platform: time_pattern
      hours: "/1"
```
It will start your automation every hour and send the weather code.

syleishere, 2025-05-11:
Yeah that what I'll have to add in each of those icon automations for now, get them triggered every hour about only way I see it working to. 

Still there are like 4 automations I have for 1 thermostat lol. This could all be easily combined in one class for sending temp and icon automatically with openweathermap, would be much appreciated especially if I consider getting 3 more of these in fall before winter hits, updating every automation again with each new ieee going to get cumbersome :)

syleishere, 2025-05-11:
Claudegel is the timeout accepted values 3600 to 18000 for both icons and outdoor temp timeouts?
When I tried playing with settings I could set anywhere from 0 to 65000 which I found strange.

claudegel, 2025-05-11:
Timeout for icons is attribute 0x0136. Timeout for outdoor temp is 0x0011. In theory both accept the same values from 0 to about 65000. In fact attr 0x0011 accept value from 1 to 65000 but for 0x0136 it accept 3600 to 18000. if you send value outside of that the button write is green but when you read it back the value is not changed.
In V1 quirks there is a specific class for the TH1134ZB-HC but in V2 quirks there is no more class. In V1 the device are identified by there signature. In V2 it is only by a combination of manufacturer/sku which is SINOPE, TH1134ZB-HC. In next update all commun attributes will be merged in sinope_base_quirk for thermostats. Only the specific attributes will be in quirkBuilder for each device. ex. the weather icons attr will be only in the TH1134ZB-HC quirkBuilder.
The advantage for the V2 is that you don't need to send data to the attribute directly and you don't need to put ieee in your automations. Actually you can make a group of thermostat (all TH1134ZB-HC) and semd the icon code to the group so you don't need to play with your automations each time you add a new device.
ex:
```
group:
  weather_icons:
    name: icons_to_thermostat
    entities:
      - select.salle_a_diner_weather_icons
      - select.bureau_weather_icons
```
then your automation will do
```
actions:
    - action: select.select_option
      data:
        option: "{{ weather_change.get(states('sensor.weather_code'), 'Hide') }}"
        entity_id: group.weather_icons
```
You can have as many group as you want. For example group all zigbee thermostat to send outdoor temp and group to send icons. In your automation you can have a trigger for temperature change and one for icon changes plus one for time pattern of each hour. Then you make your action like this:
```
actions:
    - action: select.select_option
      data:
        option: "{{ weather_change.get(states('sensor.weather_code'), 'Hide') }}"
        entity_id: group.weather_icons
    - action: number.set_value
      data:
        value: "{{ ( trigger.to_state.state|float * 100 ) |int }}"
        entity_id: group.temperature
```
your temperature group will be like this:
```
group:
  temperature:
    name: temp_to_thermostat
    entities:
      - number.salle_a_diner_outdoor_temperature
      - number.bureau_outdoor_temperature
```
So leave your timeout for outdoor temp to 10800 and for icons to 18000 and add a trigger time pattern to fire the automation each hour in case nothing change for icons or temperature.
As both icons code and outdoor temperature use a trigger state I'm not sure you can combine temperature and icon in the same automations but with group it is easy to have only two automations to call each group and manage your group when you add device.
Once I'm finish with V2 quirks I'll redo all blueprint for automations.
Also for openweathermap we cannot combine in one automation as the icon code are not the same. ex.during night, "200": Cloudrainmoon2 
durng day, "200": Cloudrainsun2
For Environment Canada there is only one automation as code are different.
Maybe it can be done but I'll play with that after I'm finish with the quirks V2
Also I'll add the missing outdoor_temp_timeout in V2

claudegel, 2025-05-11:
The local temperature offset is to fine tune your thermostat temperature sensor. Sometimes we found some device that are off a little. This is to get the correct room temperature.
The TH1134ZB-HC can be connected with a HP6000XX device to control heatpump with fan, swing etc so the quirkBuilder find the fan cluster and add automatically a fan button which is not used if you don't have a HP6000xx device. I can prevent this but it will be done at the end when all attributes are added.

syleishere, 2025-05-12:
```
conditions:
    - condition: state
      entity_id: 'sun.sun'
      state: 'above_horizon
```

If the class could call sun.sun entity and get the value, it could automatically know which icons to use for daytime or nighttime. I'm not sure if you can get the above_horizon etc value in the class, but if you could just be a simple if var == "above_horizon" use these icons, else use these icons type thing to completely automate it with openweathermap.

For a complete automation with it, could could check if entity sensor.openweathermap_temperature exists, if it does then start sending correct icons/outdoor temp type thing. Python must provide functions to check values of entities or if they exist I imagine.

Yeah thank-you for adding outdoor temp timeout, that could come in useful.

claudegel, 2025-05-13:
I've update quirks V2 branch. Added outdoor_temp_timeout