claudegel/sinope-zha #61: No angle in LM4110-ZB integration with Quirk

Christianbouchard2, 2025-01-20:
Hi, I added LM4110-ZB to Home Assistant with quirk. In integration, I only have the battery percentage and the temperature but I do not have the angle. I found it with ZHA toolkit but it's not ideal. Is it possible for you to add it in integration with the quirk, that would be good. THANK!

Christianbouchard2, 2025-03-24:
Thank you Claude for your reply. I did everything you suggested in your procedure, and it works perfectly. I have a small problem with the sensor from time to time, though. I do a scan with Zigbee Toolkit every 15 minutes. But if by chance I restart my home assistant during this scan, and it goes to 30 minutes, the sensor stops responding. I have to resynchronize it. Do you have any idea why it's doing this? I have the entire Sinopé product range, and this is the only one that does this. Yet, it also reads the temperature and battery on its own.

I can't wait to see your version 2; it might solve my problem.

Thanks again!

claudegel, 2025-03-25:
The LM4110-ZB is what we call a dormant device. To save battery it stay quiet and wake up at specific interval or when there is a change in it's measured data. If you poll too often then you will discharge the battery faster. Your LM4110-ZB is outside where it's cold and this also reduce battery life.
Next step will be to configure reporting directly for the cluster/attribute that should send the data and link them with your zigbee gateway so the LM4110-ZB will send the data by itself. Normally the LM4110-ZB send angle once per hour or if there is a change in the measured value.  
In my readme.md there is a section about reporting config of each devices. You will also find a section (for dimmer to send double clic signal) on howto send reporting configuration to your device. As it is a dormant device you need to include many retry to make sure you catch the LM4110-ZV when it wake up to send data.
In my next quirks release I'm switching everything to zha quirks V2 where it is possible to configure the reporting and create the required sensors for every devices linked to ZHA.
For now I think it is better to reduce the polling frequency to one hour. Your propane level do not go down very fast.

claudegel, 2025-04-26:
You can try the new V2 quirks in branch:
https://github.com/claudegel/sinope-zha/tree/quirks-V2
Just replace the sensor.py in your config/zhaquirks by the one in that branch and restart HA. 
For the other files, light.py is still buggy but thermostat.py and switch.py can be tested if you want.
The main difference with the quirks-V2 is that all reporting attributes are configured automatically and create a sensor associated to catch and store the reported value.
Try it and let me know.