claudegel/sinope-zha #62: LM4110-ZB Setup Feedback

patches4231, 2025-01-28:
claudegel,

Just used this to setup my new tank sensor. Thank you for your hardwork on this. Literally signed up for Paypal just to support!

One minor feedback item I had, under the Automation setup for the sensor.current_angle, I had to remove "- id: '1692841759194'" to get this to save under Automations. I think this is the Id assigned on your HA when it was created.

One question to finalize this, under the template for % calculation, is this setup as an automation? 

Thank you again.

patches4231, 2025-01-28:
I rescind my questions regarding the % Level Calculation. For those that hit this same roadblock, the template for % Canulation must be added in the config.yaml file same as the input_number and input_text

claudegel, 2025-01-28:
Yes the two input_text and input_number are placed in configuration.yaml
For the calculation they are set as a template sensor. In configuration.yaml put this
template: !include template.yaml
Then create a file called template.yaml and put the sensor in like this. You don't need to put template: at the first line like what it is in the readme.md
```
  - trigger:
    - platform: time_pattern
      hours: "/5"
  - sensor:
    - name: "Tank remaining level"
      unit_of_measurement: "%"
      unique_id: sensor.tank_remaining_level
      icon: mdi:propane-tank
      state_class: measurement
      state: >
        {% set gauge = states('input_text.tank_gauge') %} # must match the above input_text name
        {% set angle = states('sensor.current_angle') | float %}
        {% set offset = states('input_number.gauge_offset') | float %} # must match the above input_number name
        {% if (gauge == "10-80") %}
          {% set x_min = 110 | float %}
          {% set x_max = 406 | float %}
          {% set delta = 46 | float %}
          {% set low = 10 | float %}
          {% set high = 80 | float %}
        {% else %}
          {% set x_min = 120 | float %}
          {% set x_max = 415 | float %}
          {% set delta = 55 | float %}
          {% set low = 5 | float %}
          {% set high = 95 | float %}
        {% endif %}

        {% if ((delta <= angle) and (70 >= angle)) %}
          {{ (((delta-x_min)/(x_max-x_min)*(high-low))+low+offset) | round(0) }}
        {% elif ((0 <= angle) and (delta > angle)) %}
          {{ (((angle)+360-x_min)/(x_max-x_min)*(high-low)+low+offset) | round(0) }}
        {% else %}
          {{ (((angle)-x_min)/(x_max-x_min)*(high-low)+low+offset) | round(0) }}
        {% endif %}
```

patches4231, 2025-01-28:
Greatly appreciate that. That does organize it much better. Any idea why I would be getting a -13 on the output? Tank is 5-95 that was filled a couple days ago.

patches4231, 2025-01-28:
Also, while under the template i can get the % calc to provide a result but when inspecting the tank level entity it states as unavailable?

claudegel, 2025-01-28:
Did you put 
initial: "5-95" 
for the input_text ?

patches4231, 2025-01-28:
Yes i did.

I just tried modifying the delta on the 5-95 to 56 in lieu of 55 and it reads out at 98% full. This may be a dirty fix but I could see the install on my sensor itself being mis-aligned at causing this slight variation. 

I also just checked the tank level % entity and it has updated as well

closed 2025-01-28 by patches4231

claudegel, 2025-01-28:
OK perfect I'll put a note in the example in the readme.md
Thank you