claudegel/sinope-zha #63: Just Purchased the Sinope Technologies LM4110-ZB

Dhostetter, 2025-01-28:
I just purchased the Sinope Technologies LM4110-ZB sensor.
Added it to HA via ZHA without an issue.
But I don't not see a sensor for the tank reading.
I get the following sensor info:
Temperature
Battery
Identity
Not Shown:
LQI
RSSI

Did the tank level show up after adding or did you need to do something else?
Not sure what to do at this point.

Any assistance would be appreciated.

claudegel, 2025-01-29:
Hi once the LM4110ZB is installed in ZHA you need to perform many opertions in order to get the level percent. This device do not report the level % of the tank but the angle of the tank gauge needle. So you need to perform those steps:
1- setup reporting on the device
2- bind the device to your gateway so it can send the report to
3- make an automation that will get the reported angle 
4- make an input_number to set the gauge offset in configuration.yaml
5- make an input_text variable to put the gauge type you have on the tank, 10-80 or 5-95
6- make a sensor that will do the calculations and show the % level value.

Very simple. Just follow the instructions on my readme.md on this sinope-zha component
nce done you'll be able to set alarm for minimum tank level as you like

Dhostetter, 2025-01-29:
Claude,Thanks so much for responding.I'm slowly getting it together.I'm getting an error in the template.- trigger: - platform: time_pattern <- String does not match the pattern of "Legacy_Syntex^". hours: "/5"Any suggestions on how to correct this error?Get BlueMail for DesktopClaude Gelinas wrote:
Hi once the LM4110ZB is installed in ZHA you need to perform many opertions in order to get the level percent. This device do not report the level % of the tank but the angle of the tank gauge needle. So you need to perform those steps:
1- setup reporting on the device
2- bind the device to your gateway so it can send the report to
3- make an automation that will get the reported angle
4- make an input_number to set the gauge offset in configuration.yaml
5- make an input_text variable to put the gauge type you have on the tank, 10-80 or 5-95
6- make a sensor that will do the calculations and show the % level value.
Very simple. Just follow the instructions on my readme.md on this sinope-zha component
nce done you'll be able to set alarm for minimum tank level as you like

claudegel, 2025-01-30:
We need to change the trigger part like this
```
template:
  - trigger:
    - trigger: time_pattern
      hours: "/5"
```
platform should be replaced by trigger

Dhostetter, 2025-02-03:
Claudegel,
I haven't been able to get the tank level to show up. I've tried a couple of things to try and troubleshoot but nothing seems to work.
Here's where I'm at.
I'm using the second automation to retrieve the data. 
When I execute the command thru developer tools, I get this info:
zha_toolkit_version: v1.1.25
zigpy_version: 0.74.0
zigpy_rf_version: 0.13.1
ieee_org:
  - 63
  - 229
  - 171
  - 254
  - 255
  - 101
  - 17
  - 44
ieee: 2c:11:65:ff:fe:ab:e5:3f
command: attr_read
command_data: null
start_time: "2025-02-03T14:55:09.659185+00:00"
errors: []
params:
  cluster_id: 12
  attr_id: 85
  dir: 0
  tries: 100
  expect_reply: true
  args: []
  kwargs: {}
  state_id: sensor.current_angle
  allow_create: true
  read_before_write: true
  read_after_write: true
attr_type: "0x39"
write_is_equal: false
result_read:
  - "85": 351
  - {}
success: true

I've added the input gauge_offset, tank_gauge, & tank remaining level template to my config yml file.
All entities show in HA helpers. The guage_offset doesn't show under HA entities. 
When I look at the template entity its status is either not provided or unavailable.

Any suggestions on how to troubleshoot this?

claudegel, 2025-02-04:
I've noticed that there was an indentation error in the second automation for the trigger.
```
- id: '1692841759194'
  alias: Lecture niveau propane
  description: Lecture du niveau de réservoir à chaque heure
  triggers:
    - trigger: time_pattern
      hours: /1
  condition: []
  actions:
    action:
      - service: zha_toolkit.execute
        data:
          command: attr_read
          ieee: "«your LM4110 ieee »"
          cluster: 12
          attribute: 85
          state_id: sensor.current_angle
          allow_create: true
          tries: 100
  mode: single
```
did you create your input_number and input_text in configuration.yaml ?

claudegel, 2025-02-04:
I think you can remove the trigger part in the template sensor.
remove:
```
  - triggers:
      - trigger: time_pattern
        hours: "/5"
```
it should recalculate each time your sensor.current_angle is updated

Dhostetter, 2025-02-04:
Claudegel,Sorry for being a pain about this.I've attached my configuration and template files.It has to be something in the files but I just don't see it.If you have time take a look and let me know if you have anymore suggestions.As always, thanks.Get BlueMail for DesktopClaude Gelinas wrote:
I think you can remove the trigger part in the template sensor.
remove: - triggers: - trigger: time_pattern hours: "/5" it should recalculate each time your sensor.current_angle is updated

claudegel, 2025-02-04:
Sorry but I don't see your files

claudegel, 2025-02-05:
OK you put the template sensor in your configuration.yaml. It's missing the template: keyword
The best thing to do to keep thing in order is to add this in your configuration.yaml
```
group: !include groups.yaml
automation mine: !include_dir_merge_list automation/
script: !include scripts.yaml
scene: !include scenes.yaml
sensor: !include sensor.yaml
command_line: !include command.yaml
timer: !include timer.yaml
template: !include template.yaml
```
for automations you have to create a directory under config, config/automation
automation mine: !include_dir_merge_list automation/ this line will tell HA to read all files *.yaml under that directory. You can put there all files you need to organize your automations. 

For template, you create a file in config, template.yaml
template: !include template.yaml is telling to HA to include that file when it start

In template.yaml put this: (check the indentation. First line is empty, next line start with two space)

```

  - sensor:
      - name: "Tank remaining level"
        unit_of_measurement: "%"
        unique_id: sensor.tank_remaining_level
        icon: mdi:propane-tank
        icon: mdi:propane-tank
        state_class: measurement
        state: >
          {% set gauge = states('input_text.tank_gauge') %}
          {% set offset = states('input_number.gauge_offset') | float %}
          {% set angle = state_attr('sensor.neviweb130_sensor_moniteur_propane', 'gauge_angle') | float %}
          {% set x_min = 110 | float %}
          {% if (gauge == "10-80") %}
            {% set x_max = 406 | float %}
            {% set delta = 46 | float %}
            {% set low = 10 | float %}
            {% set high = 80 | float %}
          {% else %}
            {% set x_max = 415 | float %}
            {% set delta = 55 | float %}
            {% set low = 5 | float %}
            {% set high = 95 | float %}
          {% endif %}

          {% if ((delta <= angle) and (70 >= angle)) %}
            {{ (((delta-x_min)/(x_max-x_min)*(high-low))+low+offset) | round(0) }}
          {% elif ((0 <= angle) and (delta > angle)) %}
            {{ (((angle)+360-x_min)/(x_max-x_min)*(high-low)+low+offset) | round(0) }}
          {% else %}
            {{ (((angle)-x_min)/(x_max-x_min)*(high-low)+low+offset) | round(0) }}
          {% endif %}  
```
This will create a template sensor so it will be treated as a template and calculated correctly.

claudegel, 2025-02-05:
Last things, when you post code, to keep the indentation you need to paste your code then select the pasted code and click on the icon <> to keep the formatting

Dhostetter, 2025-02-05:
Claudegel,
Thanks for your help. At first it didn't work but then I noticed:
`{% set angle = state_attr('sensor.neviweb130_sensor_moniteur_propane', 'gauge_angle') | float %}`
I removed the neviweb130 info and it worked.
Thanks so much!

Dave

claudegel, 2025-02-05:
Sorry my mistake, I copy paste my own automation I was testing on neviweb130.
If you notice that the % is off a little with what you have on the tank you can play with {% set delta = 55 | float %} and change value for 56. It could be better for the 5-95 gauge.