claudegel/sinope-zha #70: not a valid value for dictionary value

ericlaflamme, 2025-08-30:
J'ai relu multiple fois comment configurer le tout mais je n'arrive pas a le faire fonctionner.  J'ai fait les blueprints and choisi les entity_id de mes thermostat mais semble que quelque chose m'échappe. Merci de votre aide et merci pour tout le travail que vous avez fait pour Sinope HA Mr. Claude.

`2025-08-30 15:05:11.605 ERROR (MainThread) [homeassistant.components.automation.sinope_outdoor_temp] Sinope Outdoor Temp: Repeat at step 1: Error executing script. Invalid data for call_service at pos 1: not a valid value for dictionary value @ data['ieee']
2025-08-30 15:05:11.606 ERROR (MainThread) [homeassistant.components.automation.sinope_outdoor_temp] Sinope Outdoor Temp: Error executing script. Invalid data for repeat at pos 1: not a valid value for dictionary value @ data['ieee']
2025-08-30 15:05:11.606 ERROR (MainThread) [homeassistant.components.automation.sinope_outdoor_temp] Error while executing automation automation.sinope_outdoor_temp: not a valid value for dictionary value @ data['ieee']`

claudegel, 2025-08-30:
Bonjour, je suis malheureusement en vacance et je n'ai ps acces à mon git sauf par courriel. L'erreur semble venir de votre ieee pour chaque appareils si vous en avez plusieurs.Svp inxcrire votre automatisation que je puisse la valider.Voici un exemple de ieee, 50:0b:91:40:00:03:db:c2, vous trouverez les votre dans parametre/zha, pour chacun de vos appareils.

claudegel, 2025-08-30:
Etes-vous sur le ZHA de base de HA. Si oui vous devriez changer pour sinope-zha. Installez la branche quirks V2. Les automatisations sont bien plus faciles. Simplement suivre les instructions du readme.md de la page d'accueil pour l'installation.

claudegel, 2025-08-31:
Pour utiliser le quirk-V2. Tu dois avoir dans la configuration de zha custom_quirks_path: /config/zhaquirks/Me rapelle pas si on met un / a la fin.Ensuite dans /config tu crès un sou-rèpertoire zhaquirks et tu copis dedans les 4 fichiers de ma branche quirks-V2; thermostat.py, light.py, sensor.py et switch.py.Ensuite redémarres HA. Ça va remplacer les quirks de sinopè dans ZHA.Ensuite tu va dans parametres / zha (zigbee home automation), selectionne ton thermostat et tu vas voir qu'il y a plein de nouveau attributs de configuration. Il y en a un pour la temperature exterieure Outdoor temp. Clic dessus et tu va trouver le nom de cet entite qui est number.ton_thermostat_outdoor_tempTu peux faire une automation qui envois la temperature exterieure a ton thermostat en l'envoyant a cet entité. La temperature doit etre envoyee en centieme de degre C ex. 10.5 oC tu envois 10.5 x 100 = 1050Tu peux le tester en ecrivant directement le chiffre a cote de Outdoor_temp sur la page du thermostat dans zha.

claudegel, 2025-09-01:
Si tu as plusieurs thermostats tu peux creer un groupe dans HA qui contient tout tes number.ton thermosta_outdoor_tempEnsuite tu te fais une automation qui envoi la temperature exterieur au groupe. Ha se charge de l'envoyer a chaque membre du groupe.Plus besoin de se preoccuper des ieee

ericlaflamme, 2025-09-04:
Merci, j'ai finalement fait fonctionné la temperature sur mes sinope. En effet, l'option apparaissait de mes devices comme mentionné. J'ai fait un template pour multiplier par 100 pour ensuite faire une automatisation de setter cette entity. Je ne sais pas pourquoi mais il semble que peut importe ce qui est entré dans ce champs, il va aprait par 0.5 sur l'appareil. Example si je met 1840, il va apparaitre 18.5 sur le sinope.

Voici mes configurations:

```
- id: '1756688523066'
  alias: Automation Sinope Outdoor Temp
  description: ''
  triggers:
  - entity_id:
    - sensor.outdoor_temp_sinope_100
    trigger: state
  conditions: []
  actions:
  - action: number.set_value
    data:
      value: '{{ outdoor_temp_100 }}'
    target:
      entity_id:
      - number.sinope_chambre_enfants_outdoor_temperature
      - number.sinope_chambre_principale_outdoor_temperature
  variables:
    outdoor_temp_100: '{{ states (''sensor.outdoor_temp_sinope_100'') | int }}'
``` 

Template/Helpers sensor:

```
{{states ('sensor.longueuil_temperature')| float * 100  }}
``` 

Merci de votre aide!

claudegel, 2025-09-04:
Les thermostats affichent la temperature en l'arrondissant à 0.5 prèsSi tu envois 1842, il va afficher 18.51822 il va afficher 18.0.Je suis de retour demain et je vais valider